LuckfoxTECH/luckfox-pico #1: Is Luckfox pico chip board support wifi?

archichen, 2023-08-17:


Luckfox01, 2023-08-18:
hi,archichen
Not supported for now, but a version with wifi may be released later.

epenli, 2023-08-26:
when?

luckfox-eng33, 2023-09-07:
> when?

Not so fast. It could be months

bonuschild, 2023-10-09:
> > when?
> 
> Not so fast. It could be months

So great! That's so nice. Will be better than raspberry pi zero then!

versamodule, 2023-11-03:
I got my pico in today. I think it's great!  But I have to say there is 2 major shortcomings that I think would have made it one of the most popular boards out there even surpassing the pi. 
#1) Wi-Fi should have been onboard, or at the very least you should have added pins for USB, so at least we could use a Wi-Fi dongle. Without internet connectivity it seriously hinders it.
#2) I expected an OS more like the pi, or what dietPi offers (still a newb to Linux so not sure what to call it) Basic things like apt-get  update and such.

gehee, 2024-01-06:
Can’t the usb port be used to plug a WiFi dongle? If not, we definitely need a version with usb pins. That’d be a game changer

archichen, 2024-01-21:
> I got my pico in today. I think it's great! But I have to say there is 2 major shortcomings that I think would have made it one of the most popular boards out there even surpassing the pi. #1) Wi-Fi [...]

Indeed. @Luckfox01 Hi team, have any plan for these features?  I think a chip board that is small enough but power enough is very interesting.

luckfox-eng33, 2024-01-22:
We will soon release a version with wifi and Bluetooth.
May be released before Chinese New Year.
But we do not rule out that it will be after Chinese New Year.

andrethomas, 2024-02-10:
@luckfox-eng33 any possibility on using ESP8266 as a wifi adapter, for example [https://hackaday.io/project/9300/instructions](https://hackaday.io/project/9300/instructions)

meetroger, 2024-02-24:
Any updates? What chipset will be supported first?

versamodule, 2024-02-24:
Have you guys seen these? 
https://www.cnx-software.com/2024/02/08/licheerv-nano-low-cost-sg2002-risc-v-arm-camera-display-board-wifi-6-ethernet/

vjoco, 2024-03-09:
it sure does, I'm using usb wifi adapters! Not easy but doable to compile the kernel with driver support, set the wpa_supplicant and networking!

versamodule, 2024-03-09:
@vjoco,  where are you getting the USB signals from for the wifi adapter?

andrethomas, 2024-03-09:
You can change the mode of the USB port to host mode as per [wiki](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-USB) but am also curious about how @vjoco is wiring things up because that's also the primary way of powering the board... perhaps feeding power in via the VBUS pins 40 (5V) and pin 38 (GND) ?

versamodule, 2024-03-09:
Either that, or a USB hub.

versamodule, 2024-03-09:
That is how I am powering mine. I made a POE version PCB for it. 

[image]

andrethomas, 2024-03-09:
That looks really neat. I am yet to get my hands on the mini ones... only have a pro max at the moment but I also prefer the spi-nand over sd card so likely I'll need to add or upgrade the spi-nand when I eventually get my hands on a mini.

vjoco, 2024-03-09:
@andrethomas  and @versamodule 
I'm using a pico mini a, with an external 5V supply on vbus and gnd (pins 1 and 2). The usb is configured as usb host according with the wiki, and i use an otg type c to type a adapter to connect peripherals. 
[image]
 I even attached an usb hub and i connected an extra usb cam, and even an usb to serial adapter.

 I think there should be no problem to use 5V on Vbus on any of the luckfox version to have a powered usb host. Pico mini a works just great.

vjoco, 2024-03-09:
the 3 wifi sticks I'm using are based on rtl8192cu and mt7601.. whatever usb to wifi sticks are supperted by the kernel, they will work.

andrethomas, 2024-03-09:
That's great, thanks for sharing. I'm looking at getting some of these for my usb experiments instead of using a "normal" usb hub so that'll work out well for adding an extra usb serial console

[https://www.aliexpress.com/item/1005001596841814.html](https://www.aliexpress.com/item/1005001596841814.html)

The Pro Max has TP1 and TP2 pads on the bottom of the board which I think may be connected to the USB D- and D+ pins so that'll be interesting.... not sure if there's room on the mini's for that kind of connection though but I guess a usb-c adapter that has solder pads on it will also work fine if the TP1 and TP2 turn out to NOT be D+ and D-

versamodule, 2024-03-09:
@vjoco, thanks for the update. I too am using the c to type a adapter. I changed the USB mode to host mode.

&usbdrd_dwc3 {
status = "okay";
dr_mode = "host";
};

Recompiled, and uploaded to board. 

Just barley know enough to get by in Linux, but where I am stuck at is how to set the wpa_supplicant and networking stuff. 
Since the terminal is not the typical one i see for like Raspberry Pi.

vjoco, 2024-03-09:
@andrethomas , the hub you just shared (aliexpress) is great! For the mini a, maybe there is some usb-c adapter with the D+ D- lines to be wired to that

vjoco, 2024-03-09:
@versamodule 
are you seeing the device with lsusb? check ohci/ehci support in kernel.
paste the kernel messages (from dmesg) . You may need to have the firmware bin files in /lib/firmware
wpa_supplicant will be you next fight but don't give up. Networking is quite easy.
Are you using buildroot for rootfs?

andrethomas, 2024-03-09:
@vjoco indeed, I see these kind of connectors which seems to be for if you want to make your own usb-c cable so these should work well for a short connection to a mini without using the "shell" part of the assembly... and then I can feed 5V in through the usb-c port also!

[image]

versamodule, 2024-03-09:
@vjoco, using lsusb i do see the device. 
>Are you using buildroot for rootfs?
Sad to say, I have no clue. I was just following the instructions on how to build it after making the changes. 

> [    0.111702] i2c /dev entries driver
[    0.113748] i2c i2c-4: 1 i2c clients have been registered at 0x30
[    0.114146] i2c i2c-4: 2 i2c clients have been registered at 0x30
[    0.116348] rockchip-thermal ff3c8000.tsadc: tsadc is probed successfully!
[    0.116883] cpu cpu0: Failed to get cpu_leakage
[    0.117739] cpu cpu0: avs=0
[    0.117922] cpu cpu0: opp calibration done
[    0.118053] cpu cpu0: failed to register system monitor
[    0.118514] Synopsys Designware Multimedia Card Interface Driver
[    0.119483] dwmmc_rockchip ffaa0000.mmc: IDMAC supports 32-bit address mode.
[    0.119536] dwmmc_rockchip ffaa0000.mmc: Using internal DMA controller.
[    0.119554] dwmmc_rockchip ffaa0000.mmc: Version ID is 270a
[    0.119602] dwmmc_rockchip ffaa0000.mmc: DW MMC controller at irq 51,32 bit host data width,256 deep fifo
[    0.122947] rv1106-acodec ff480000.acodec: current soc_id: rv1103
[    0.123071] rv1106-acodec ff480000.acodec: Use pa_ctl_gpio and pa_ctl_delay_ms: 0
[    0.124977] NET: Registered protocol family 17
[    0.125102] Key type dns_resolver registered
[    0.126067] Loading compiled-in X.509 certificates
[    0.126779]
[    0.126816] ********************************************************************
[    0.126826] **     NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE           **
[    0.126833] **                                                                **
[    0.126838] **  WRITEABLE clk DebugFS SUPPORT HAS BEEN ENABLED IN THIS KERNEL **
[    0.126844] **                                                                **
[    0.126850] ** This means that this kernel is built to expose clk operations  **
[    0.126856] ** such as parent or rate setting, enabling, disabling, etc.      **
[    0.126862] ** to userspace, which may compromise security on your system.    **
[    0.126867] **                                                                **
[    0.126872] ** If you see this message and you are not debugging the          **
[    0.126879] ** kernel, report this immediately to your vendor!                **
[    0.126884] **                                                                **
[    0.126890] **     NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE           **
[    0.126895] ********************************************************************
[    0.132504] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    0.146417] input: adc-keys as /devices/platform/adc-keys/input/input0
[    0.214077] ubi0: attaching mtd6
[    0.358981] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[    0.381475] ubi0: scanning is finished
[    0.386313] ubi0: attached mtd6 (name "rootfs", size 80 MiB)
[    0.386341] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[    0.386350] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[    0.386359] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
[    0.386367] ubi0: good PEBs: 640, bad PEBs: 0, corrupted PEBs: 0
[    0.386374] ubi0: user volume: 1, internal volumes: 1, max. volumes count: 128
[    0.386393] ubi0: max/mean erase counter: 2/0, WL threshold: 4096, image sequence number: 447760338
[    0.386402] ubi0: available PEBs: 0, total reserved PEBs: 640, PEBs reserved for bad PEB handling: 20
[    0.387417] ALSA device list:
[    0.387452]   #0: rv-acodec
[    0.388103] ubi0: background thread "ubi_bgt0d" started, PID 53
[    0.388376] UBIFS (ubi0:0): Mounting in unauthenticated mode
[    0.405157] UBIFS (ubi0:0): recovery needed
[    0.427005] UBIFS (ubi0:0): recovery deferred
[    0.427137] UBIFS (ubi0:0): UBIFS: mounted UBI device 0, volume 0, name "rootfs", R/O mode
[    0.427156] UBIFS (ubi0:0): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[    0.427168] UBIFS (ubi0:0): FS size: 76820480 bytes (73 MiB, 605 LEBs), journal size 9023488 bytes (8 MiB, 72 LEBs)
[    0.427177] UBIFS (ubi0:0): reserved for root: 0 bytes (0 KiB)
[    0.427188] UBIFS (ubi0:0): media format: w4/r0 (latest is w5/r0), UUID C8F1FA18-99D9-44BA-8824-83CE855B1DD7, small LPT model
[    0.428285] VFS: Mounted root (ubifs filesystem) readonly on device 0:14.
[    0.429123] devtmpfs: mounted
[    0.429316] Freeing unused kernel memory: 160K
[    0.429326] Kernel memory protection not selected by kernel config.
[    0.429363] Run /sbin/init as init process
[    0.429373]   with arguments:
[    0.429377]     /sbin/init
[    0.429380]   with environment:
[    0.429384]     HOME=/
[... 135 lines cut ...]
[    2.787145] mpp_vcodec: loading out-of-tree module taints kernel.
[    2.805731] mpp_vcodec: init new
[    2.806682] mpp_service mpp-srv: 3760314f author: Zhihua Wang 2023-01-12 [vcodec]: check chan create when destroy
[    2.806705] mpp_service mpp-srv: probe start
[    2.809066] mpp_rkvenc_540c ffa50000.rkvenc: probing start
[    2.810882] mpp_rkvenc_540c ffa50000.rkvenc: probing finish
[    2.811391] mpp_vepu_pp ffa60000.rkvenc-pp: probe device
[    2.811512] mpp_vepu_pp ffa60000.rkvenc-pp: shared_rst_a_pp is not found!
[    2.811525] vepu_pp_init:528: No aclk reset resource define
[    2.811535] mpp_vepu_pp ffa60000.rkvenc-pp: shared_rst_h_pp is not found!
[    2.811542] vepu_pp_init:531: No hclk reset resource define
[... cut ...]
[    2.847110] RKNPU ff660000.npu: RKNPU: rknpu iommu device-tree entry not found!, using non-iommu mode
[    2.848551] RKNPU ff660000.npu: RKNPU: Initialized RKNPU driver: v0.8.2 for 20220829
[    2.848674] RKNPU ff660000.npu: dev_pm_opp_set_regulators: no regulator (rknpu) found: -19
[    3.139773] rockit_load start
[    3.156343] rockit_load end
[    3.289526] file system registered
[    3.351226] using random self ethernet address
[    3.351265] using random host ethernet address
[    3.709543] read descriptors
[    3.709730] read strings

>

vjoco, 2024-03-10:
@versamodule 
See the kernel support for the usb wifi, I documented briefly on this entry:
https://github.com/LuckfoxTECH/luckfox-pico/issues/78

I think the dmesg you pasted is missing the last part, ... or since the support for the usb wifi is not included, the kernel does not even try to assign a driver.

What's the output of lsusb?

versamodule, 2024-03-11:
Thank for the help!  That was the entire dmesg that I posted. 
lsusb results are: 

# lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 0bda:8176
Bus 002 Device 001: ID 1d6b:0003
#

If I unplug the WiFi I get: 
# [   57.499297] usb 1-1: USB disconnect, device number 2

When I plug it in I get:
# [   97.189675] usb 1-1: new high-speed USB device number 3 using xhci-hcd
[   97.519864] usb 1-1: device descriptor read/64, error -71
[   97.767790] usb 1-1: New USB device found, idVendor=0bda, idProduct=8176, bcdDevice= 2.00
[   97.767851] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   97.767877] usb 1-1: Product: 802.11n WLAN Adapter
[   97.767901] usb 1-1: Manufacturer: Realtek
[   97.767925] usb 1-1: SerialNumber: 00e04c000001

I never tried "build.sh kernel", I have only used "sudo ./build.sh"  I will give what you suggested a try.

versamodule, 2024-03-12:
@vjoco
I truly appreciate you taking the time to help.  I do not have wlan0.  
I am just over my head with knowing what exactly to enable and some things don't make any sense like here https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-USB/#kernel-configuration
it says to cp ./arch/arm/configs/luckfox_rv1106_linux_defconfig .config.   But I am using the RV1103.

vjoco, 2024-03-18:
@versamodule 
:) don't give up... the luckfox_rv1106_linux_defconfig is the right one

PJ5168, 2024-04-17:
This one has CSI, DSI, eMMC, and WiFi.
https://www.aliexpress.com/item/1005006782579582.html

Also, the SG2002 has WiFi as well.
https://www.aliexpress.com/item/1005006519668532.html

luckfox-eng33, 2024-04-17:
> This one has CSI, DSI, eMMC, and WiFi. https://www.aliexpress.com/item/1005006782579582.html
> 
> Also, the SG2002 has WiFi as well. https://www.aliexpress.com/item/1005006519668532.html

It's not that we dislike the authors of this board, they are not developer friendly.
And to some extent, the technical support it should bear will be transferred to our side to some extent.
So we will delete this comment

giatorta, 2024-05-01:
@luckfox-eng33 any news about wifi/BT on-board support?
we'd be very interested in a out-of-the-box HW/SW support for wireless communication
thanks!

nopnop2002, 2024-05-19:
I'm also waiting for Luckfox with WiFi.

If you have an ES8266 (ESP01), you can use it as a WiFi modem.

https://github.com/nopnop2002/python-esp8266

nopnop2002, 2024-06-06:
LuckFox Pico Ultra/Ultra W will be released.

Ultra W has WiFi.

https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPIO

giatorta, 2024-06-06:
> LuckFox Pico Ultra/Ultra W will be released.

hope it'll be soon! thank you @nopnop2002 for the info!

nopnop2002, 2024-06-06:
We can obtain multi-core products such as OrangePiZeo3 and NanoPiNeo at low cost.
Since LuxkFox is a single core, it must be available cheaper than these.

PJ5168, 2024-06-07:
It's too expensive. Will go with RV1126/1109 or RK3566 for much higher CPU and NPU performance.
-->> https://www.aliexpress.com/item/1005007133022095.htm

nopnop2002, 2024-06-07:
@PJ5168 

This is my actual measured value.

- Orangepi PC
```
========================================================================
   BYTE UNIX Benchmarks (Version 5.1.3)

   System: orangepipc: GNU/Linux
   OS: GNU/Linux -- 4.19.20-sunxi -- #5.75 SMP Sat Feb 9 19:02:47 CET 2019
   Machine: armv7l (unknown)
   Language: en_US.utf8 (charmap="UTF-8", collate="UTF-8")
   CPU 0: ARMv7 Processor rev 5 (v7l) (0.0 bogomips)

   CPU 1: ARMv7 Processor rev 5 (v7l) (0.0 bogomips)

   CPU 2: ARMv7 Processor rev 5 (v7l) (0.0 bogomips)

   CPU 3: ARMv7 Processor rev 5 (v7l) (0.0 bogomips)

   01:08:00 up 1 min,  1 user,  load average: 0.74, 0.31, 0.11; runlevel 2019-02-16

------------------------------------------------------------------------
Benchmark Run: Sun Feb 17 2019 01:08:00 - 01:36:26
4 CPUs in system; running 1 parallel copy of tests

Dhrystone 2 using register variables        4896337.2 lps   (10.0 s, 7 samples)
Double-Precision Whetstone                      839.2 MWIPS (10.0 s, 7 samples)
Execl Throughput                                815.6 lps   (29.9 s, 2 samples)
File Copy 1024 bufsize 2000 maxblocks         92320.1 KBps  (30.0 s, 2 samples)
File Copy 256 bufsize 500 maxblocks           28103.5 KBps  (30.0 s, 2 samples)
File Copy 4096 bufsize 8000 maxblocks        229999.0 KBps  (30.0 s, 2 samples)
Pipe Throughput                              201588.7 lps   (10.0 s, 7 samples)
Pipe-based Context Switching                  44489.6 lps   (10.0 s, 7 samples)
Process Creation                               1755.8 lps   (30.0 s, 2 samples)
Shell Scripts (1 concurrent)                   1670.9 lpm   (60.0 s, 2 samples)
Shell Scripts (8 concurrent)                    454.3 lpm   (60.1 s, 2 samples)
System Call Overhead                         569885.8 lps   (10.0 s, 7 samples)

System Benchmarks Index Values               BASELINE       RESULT    INDEX
Dhrystone 2 using register variables         116700.0    4896337.2    419.6
Double-Precision Whetstone                       55.0        839.2    152.6
Execl Throughput                                 43.0        815.6    189.7
File Copy 1024 bufsize 2000 maxblocks          3960.0      92320.1    233.1
File Copy 256 bufsize 500 maxblocks            1655.0      28103.5    169.8
File Copy 4096 bufsize 8000 maxblocks          5800.0     229999.0    396.6
Pipe Throughput                               12440.0     201588.7    162.0
Pipe-based Context Switching                   4000.0      44489.6    111.2
Process Creation                                126.0       1755.8    139.4
Shell Scripts (1 concurrent)                     42.4       1670.9    394.1
Shell Scripts (8 concurrent)                      6.0        454.3    757.2
System Call Overhead                          15000.0     569885.8    379.9
                                                                   ========
System Benchmarks Index Score                                         248.1
```

- Luckfox Pico max
```

========================================================================
   BYTE UNIX Benchmarks (Version 5.1.3)

   System: luckfox: GNU/Linux
   OS: GNU/Linux -- 5.10.110 -- #1 Fri Dec 29 14:34:18 HKT 2023
   Machine: armv7l (armv7l)
   Language: en_US.utf8 (charmap="ANSI_X3.4-1968", collate="ANSI_X3.4-1968")
   CPU 0: ARMv7 Processor rev 5 (v7l) (69.8 bogomips)

   07:31:27 up 14 min,  3 users,  load average: 0.38, 0.25, 0.15; runlevel Nov

------------------------------------------------------------------------
Benchmark Run: Thu May 02 2024 07:31:27 - 07:59:39
1 CPU in system; running 1 parallel copy of tests

Dhrystone 2 using register variables        4730955.0 lps   (10.0 s, 7 samples)
Double-Precision Whetstone                     1088.2 MWIPS (10.0 s, 7 samples)
Execl Throughput                                450.1 lps   (29.9 s, 2 samples)
File Copy 1024 bufsize 2000 maxblocks        100939.8 KBps  (30.0 s, 2 samples)
File Copy 256 bufsize 500 maxblocks           33634.0 KBps  (30.0 s, 2 samples)
File Copy 4096 bufsize 8000 maxblocks        255346.5 KBps  (30.0 s, 2 samples)
Pipe Throughput                              352454.7 lps   (10.0 s, 7 samples)
Pipe-based Context Switching                  87081.9 lps   (10.0 s, 7 samples)
Process Creation                               1506.1 lps   (30.0 s, 2 samples)
Shell Scripts (1 concurrent)                   1186.0 lpm   (60.0 s, 2 samples)
Shell Scripts (8 concurrent)                    154.1 lpm   (60.3 s, 2 samples)
System Call Overhead                         829129.9 lps   (10.0 s, 7 samples)

System Benchmarks Index Values               BASELINE       RESULT    INDEX
Dhrystone 2 using register variables         116700.0    4730955.0    405.4
Double-Precision Whetstone                       55.0       1088.2    197.8
Execl Throughput                                 43.0        450.1    104.7
File Copy 1024 bufsize 2000 maxblocks          3960.0     100939.8    254.9
File Copy 256 bufsize 500 maxblocks            1655.0      33634.0    203.2
File Copy 4096 bufsize 8000 maxblocks          5800.0     255346.5    440.3
Pipe Throughput                               12440.0     352454.7    283.3
Pipe-based Context Switching                   4000.0      87081.9    217.7
Process Creation                                126.0       1506.1    119.5
Shell Scripts (1 concurrent)                     42.4       1186.0    279.7
Shell Scripts (8 concurrent)                      6.0        154.1    256.9
System Call Overhead                          15000.0     829129.9    552.8
                                                                   ========
System Benchmarks Index Score                                         248.7
```

PJ5168, 2024-06-08:
@nopnop2002 
Seems that your testing results are wrong. You may need to check your program setting.
Did you check all four cores in H3 are at full load? ?? ???
The Orange Pi PC use H3 chip which is running at 1.37Ghz.
The RV1106G3 is running at 1.304Ghz.
They both use same A7 design.
Even for single core, the "Coremark" result of H3 is 8.11% higher than RV1106G3.
For all four cores, H3 is much faster than RV1106G3.
BTW, H3 uses very old process. So, it is very hot. But RV1106G3 use new process which is not hot under full load.

PJ5168, 2024-06-08:
@nopnop2002 
For cheap one, the Zero 3 may be a good choice.
-->> https://www.aliexpress.com/item/1005006170910291.html

nopnop2002, 2024-06-08:
@PJ5168 

All results of OrangePiPC's UnixBench are below.

```
========================================================================
   BYTE UNIX Benchmarks (Version 5.1.3)

   System: orangepipc: GNU/Linux
   OS: GNU/Linux -- 4.19.20-sunxi -- #5.75 SMP Sat Feb 9 19:02:47 CET 2019
   Machine: armv7l (unknown)
   Language: en_US.utf8 (charmap="UTF-8", collate="UTF-8")
   CPU 0: ARMv7 Processor rev 5 (v7l) (0.0 bogomips)

   CPU 1: ARMv7 Processor rev 5 (v7l) (0.0 bogomips)

   CPU 2: ARMv7 Processor rev 5 (v7l) (0.0 bogomips)

   CPU 3: ARMv7 Processor rev 5 (v7l) (0.0 bogomips)

   01:08:00 up 1 min,  1 user,  load average: 0.74, 0.31, 0.11; runlevel 2019-02-16

------------------------------------------------------------------------
Benchmark Run: Sun Feb 17 2019 01:08:00 - 01:36:26
4 CPUs in system; running 1 parallel copy of tests

Dhrystone 2 using register variables        4896337.2 lps   (10.0 s, 7 samples)
Double-Precision Whetstone                      839.2 MWIPS (10.0 s, 7 samples)
Execl Throughput                                815.6 lps   (29.9 s, 2 samples)
File Copy 1024 bufsize 2000 maxblocks         92320.1 KBps  (30.0 s, 2 samples)
File Copy 256 bufsize 500 maxblocks           28103.5 KBps  (30.0 s, 2 samples)
File Copy 4096 bufsize 8000 maxblocks        229999.0 KBps  (30.0 s, 2 samples)
Pipe Throughput                              201588.7 lps   (10.0 s, 7 samples)
Pipe-based Context Switching                  44489.6 lps   (10.0 s, 7 samples)
Process Creation                               1755.8 lps   (30.0 s, 2 samples)
Shell Scripts (1 concurrent)                   1670.9 lpm   (60.0 s, 2 samples)
Shell Scripts (8 concurrent)                    454.3 lpm   (60.1 s, 2 samples)
System Call Overhead                         569885.8 lps   (10.0 s, 7 samples)

System Benchmarks Index Values               BASELINE       RESULT    INDEX
Dhrystone 2 using register variables         116700.0    4896337.2    419.6
Double-Precision Whetstone                       55.0        839.2    152.6
Execl Throughput                                 43.0        815.6    189.7
File Copy 1024 bufsize 2000 maxblocks          3960.0      92320.1    233.1
File Copy 256 bufsize 500 maxblocks            1655.0      28103.5    169.8
File Copy 4096 bufsize 8000 maxblocks          5800.0     229999.0    396.6
Pipe Throughput                               12440.0     201588.7    162.0
Pipe-based Context Switching                   4000.0      44489.6    111.2
Process Creation                                126.0       1755.8    139.4
Shell Scripts (1 concurrent)                     42.4       1670.9    394.1
Shell Scripts (8 concurrent)                      6.0        454.3    757.2
System Call Overhead                          15000.0     569885.8    379.9
                                                                   ========
System Benchmarks Index Score                                         248.1

------------------------------------------------------------------------
Benchmark Run: Sun Feb 17 2019 01:36:26 - 02:04:59
4 CPUs in system; running 4 parallel copies of tests

Dhrystone 2 using register variables       19635082.2 lps   (10.0 s, 7 samples)
Double-Precision Whetstone                     3363.4 MWIPS (10.0 s, 7 samples)
Execl Throughput                               1942.8 lps   (29.9 s, 2 samples)
File Copy 1024 bufsize 2000 maxblocks        155408.2 KBps  (30.0 s, 2 samples)
File Copy 256 bufsize 500 maxblocks           46159.5 KBps  (30.0 s, 2 samples)
File Copy 4096 bufsize 8000 maxblocks        401294.6 KBps  (30.0 s, 2 samples)
Pipe Throughput                              783555.4 lps   (10.0 s, 7 samples)
Pipe-based Context Switching                 179113.8 lps   (10.0 s, 7 samples)
Process Creation                               3959.9 lps   (30.0 s, 2 samples)
Shell Scripts (1 concurrent)                   3654.7 lpm   (60.1 s, 2 samples)
Shell Scripts (8 concurrent)                    475.3 lpm   (60.4 s, 2 samples)
System Call Overhead                        2211563.9 lps   (10.0 s, 7 samples)

System Benchmarks Index Values               BASELINE       RESULT    INDEX
Dhrystone 2 using register variables         116700.0   19635082.2   1682.5
Double-Precision Whetstone                       55.0       3363.4    611.5
Execl Throughput                                 43.0       1942.8    451.8
File Copy 1024 bufsize 2000 maxblocks          3960.0     155408.2    392.4
File Copy 256 bufsize 500 maxblocks            1655.0      46159.5    278.9
File Copy 4096 bufsize 8000 maxblocks          5800.0     401294.6    691.9
Pipe Throughput                               12440.0     783555.4    629.9
Pipe-based Context Switching                   4000.0     179113.8    447.8
Process Creation                                126.0       3959.9    314.3
Shell Scripts (1 concurrent)                     42.4       3654.7    862.0
Shell Scripts (8 concurrent)                      6.0        475.3    792.2
System Call Overhead                          15000.0    2211563.9   1474.4
                                                                   ========
System Benchmarks Index Score                                         618.6
```

PJ5168, 2024-06-09:
@nopnop2002 

1. The hardware design of H618 is OK, but the software is not. That's why prefer Rockchip over Allwinner.

2. You compare a 2019 test result to a 2024 test result. All are different now. You may need to re-visit the H3 test based on latest Ubuntu 20.4 image released by OrangePi for a closer look.

3.  The UnixBench is an old school stuff. It may be OK for old Unix system and usage model. But it is NOT OK for today's Linux distro/build and usage model. You need to have a precise control over how you conduct the test for a meaningful result. Otherwise, you are testing the compiler rather than the system.
-->> https://www.eembc.org/techlit/datasheets/dhrystone_wp.pdf
-->> https://developer.arm.com/documentation/105958/latest/
-->> https://electronics.stackexchange.com/questions/494377/dmips-in-multi-core-systems

4.  There is NO standard procedure nor parameters how to conduct the Dhrystone or Whetstone tests.
     For different C compilers or compiling flags, the "score" will be several times different!
     Even for same compiler and flags, but for a different N size, the Whetstone result may be different as well.

5. For small board like this, even a different SD card can vary the IO performance.

6. For different Linux distro, busy-box, build-root, or even RTOS, bare-metal, the system related measurement is useless. It's not applicable to today's usage model for a specific solution to the targeted problem.

7. It's clear that your test results can not dientify the differences between two target boards. The UnixBench, again, fail the purpose of itself. Therefore, the test results are meaningless. 

8. So, morden chip use totally different measuremet items for today's workload. 
   You can refer to what Lisa Su said in the  intro of AMD Zen5. And what testing programs she mentioned in the computex last week.
   Or, you can check why Apple M4 has much better score on Geekbench 6 over Geekbench 5.
  And for smart phone, why MTK 9300 outperforms 8 Gen 3 on the selected testing items.

9. BTW, I use Dhrystone and Whetstone for several decades. I like the little RV1106G3 as well. But for a precise apple to apple comparison between two chips is very interesting today. Unless, you can do it on chip simulator.

2 cents~

nopnop2002, 2024-06-16:
Ultra has been released.
But it's expensive.
https://www.luckfox.com/Luckfox-Pico

plan44, 2024-07-06:
> it sure does, I'm using usb wifi adapters! Not easy but doable to compile the kernel with driver support, set the wpa_supplicant and networking!

What WiFi adapter / chipset are you using? Any recommendation? I have just built OpenWrt for the pico which has support for a lot of wifi dongles (but I haven't yet managed to flash the image to the right partitions yet 😵‍💫)

luckfox-eng33, 2024-07-09:
> What WiFi adapter / chipset are you using? Any recommendation? I have just built OpenWrt for the pico which has support for a lot of wifi dongles (but I haven't yet managed to flash the image to the [...]

Maybe you can refer to it or use it directly[Luckfox-Pico-Ultra-W](https://www.luckfox.com/EN-Luckfox-Pico-Ultra-W)

giatorta, 2024-09-21:
> @luckfox-eng33 any possibility on using ESP8266 as a wifi adapter, for example https://hackaday.io/project/9300/instructions

has anyone found any hints on how to exploit ESP8266 (or ESP32) as a wifi-adapter maybe via SPI/UART?
thanks in advance!

nopnop2002, 2024-09-22:
@giatorta 

Network example using pyserial and esp8266 AT firmware.
https://github.com/nopnop2002/python-esp8266

plan44, 2024-09-22:
I guess a promising solution would be ESP-Hosted NG: https://github.com/espressif/esp-hosted

luckfox-eng33, 2024-09-23:
> > @luckfox-eng33 any possibility on using ESP8266 as a wifi adapter, for example https://hackaday.io/project/9300/instructions
> 
> has anyone found any hints on how to exploit ESP8266 (or ESP32) as a wifi-adapter maybe via SPI/UART? thanks in advance!

It seems someone has already replied to you. 👍

Antom91, 2025-04-25:
What pins to connect USB data pins to luckfox pico min A ?

nopnop2002, 2025-04-25:
>What pins to connect USB data pins to luckfox pico min A ?

Use on board usb connector.
Power is supplied via the power pins.

Antom91, 2025-04-26:
> > What pins to connect USB data pins to luckfox pico min A ?
> 
> Use on board usb connector. Power is supplied via the power pins.
> 
> https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-USB/

I don't understand, that is "on board usb connector" ?

Antom91, 2025-05-01:
Okay, i have Wi-Fi USB dongle r8188eu

If i connect or reconnect this device,e I get 
```
[   57.579237] usb 1-1: USB disconnect, device number 2
[   60.615134] usb 1-1: new high-speed USB device number 3 using xhci-hcd
[   65.172110] usb 1-1: USB disconnect, device number 3
[   67.705141] usb 1-1: new high-speed USB device number 4 using xhci-hcd
[   81.405744] usb 1-1: USB disconnect, device number 4
[   83.355154] usb 1-1: new high-speed USB device number 5 using xhci-hcd
[  146.680861] usb 1-1: USB disconnect, device number 5
[  152.865123] usb 1-1: new high-speed USB device number 6 using xhci-hcd
```

if I connect USB flash drive I get

```
[   34.922744] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
[   57.579237] usb 1-1: USB disconnect, device number 2
[   60.615134] usb 1-1: new high-speed USB device number 3 using xhci-hcd
[   65.172110] usb 1-1: USB disconnect, device number 3
[   67.705141] usb 1-1: new high-speed USB device number 4 using xhci-hcd
[   81.405744] usb 1-1: USB disconnect, device number 4
[   83.355154] usb 1-1: new high-speed USB device number 5 using xhci-hcd
[  146.680861] usb 1-1: USB disconnect, device number 5
[  152.865123] usb 1-1: new high-speed USB device number 6 using xhci-hcd
[  153.052521] usb-storage 1-1:1.0: USB Mass Storage device detected
[  153.065064] scsi host0: usb-storage 1-1:1.0
[  154.086091] scsi 0:0:0:0: Direct-Access     SanDisk  Ultra Backup     8.32 PQ: 0 ANSI: 0 CCS
[  154.104462] sd 0:0:0:0: [sda] Attached SCSI removable disk
[  156.465938] sd 0:0:0:0: [sda] 31301631 512-byte logical blocks: (16.0 GB/14.9 GiB)
[  156.467238]  sda: sda1

```

Looks like USB is working! 

How to connect a Wi-Fi dongle?

giatorta, 2025-05-04:
Hi Antom, which Luckfox do you have? How did you connect the dongle (with USB 2.0) to the Luckfox?

Antom91, 2025-05-04:
> Hi Antom, which Luckfox do you have? How did you connect the dongle (with USB 2.0) to the Luckfox?

Luckfox micro mini a (or b)

You can connect 5V to the VBUS and GND pins, then use TypeC OTG cable (or connector). USB WiFI --> USB TypeC OTG Cabel --> TypeC connector on Luckfox board