LuckfoxTECH/luckfox-pico #2: How to build python or other applications for the board?

candyriver, 2023-09-02:
I want to use gcc, and python directly in the board. Could you please provide more tutorials ?

luckfox-eng33, 2023-09-07:
It's in the plan, but it's not a high priority.
It could take a few months.

bonuschild, 2023-10-09:
Can a program built as an `armV7` architecture directrly run on this board? It was built with go and cross-compile toolchain with binary size about 4MB in total.
If yes that would be OK but if not, what is the solution to this? Thank you.

luckfox-eng33, 2023-10-11:
> Can a program built as an `armV7` architecture directrly run on this board? It was built with go and cross-compile toolchain with binary size about 4MB in total. If yes that would be OK but if not, what is the solution to this? Thank you.

Maybe, you can give it a try, it is best to recompile through the toolchain we provide

bonuschild, 2023-10-11:
> Maybe, you can give it a try, it is best to recompile through the toolchain we provide

It could works! So what is the different between the toolchain you provided and the common toolchain?

luckfox-eng33, 2023-10-11:
> It could works! So what is the different between the toolchain you provided and the common toolchain?

toolchain is provided by rockchip with little or no modification

bonuschild, 2023-10-15:
Yes of course. It work and need no more modifications. Just put it on your flash storage and execute it. To be remember is that you executable may depends on other libraries. Copy them to the `/lib` directory so that your executable can find all its dependencies.
        
    Let me know if you work.
        
            
        
    
         
        
            
                        ***@***.***
            
        
     
        
  
    ---- Replied Message ----
  
  
    
      
        
         From 
        
        
            ***@***.***>
            
        
      
      
        
         Date 
        
        
        10/15/2023 10:08
        
      
      
        
         To 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         Cc 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         Subject 
        
        
              Re: [LuckfoxTECH/luckfox-pico] How to build python or other applications for the board? (Issue #2)
        
      
    
  

Can a program built as an armV7 architecture directrly run on this board? It was built with go and cross-compile toolchain with binary size about 4MB in total. If yes that would be OK but if not, what is the solution to this? Thank you.

Did it work? If it did, please give me some advice. Thanks!

closed 2024-03-01 by andrethomas

B3B3K, 2025-01-20:
actually, for luckfox pico mini (RV1103), with opkg, you can download gcc!
http://pkg.entware.net/binaries/armv7/
(download bin is in opt)(export PATH=$PATH:/opt/bin)