LuckfoxTECH/luckfox-pico #8: certificate & timezone support missed in default busybox image

bonuschild, 2023-10-10:
I use luckfox max(256MB) and run an TLS application but it report that：
- time is incorrect, it was 2021 and timezone can not be set via `date -s`. How to set the timezone of the board?
- " x509: certificate signed by unknown authority", properly because there is no ca-certificates support at all. How to add the support?

luckfox-eng33, 2023-10-11:
Changing the time zone is not supported at the moment

bornmw, 2026-09-18:
Cross-reference: issue #373 has fresh (2026-09-17) verified diagnostics on a LuckFox Pico Plus buildroot image confirming the CA part of this report: no CA bundle anywhere under /etc//usr//var, ssl.get_default_verify_paths() cafile=None with empty /etc/ssl/certs capath, system-default TLS verification of api.telegram.org fails, and a bundled IETF CA set verifies fine. See #373.