LuckfoxTECH/luckfox-pico #9: partition and memory size problem

bonuschild, 2023-10-10:
I use a **luckfox max** with 256MB flash. But when `df -h` it give these results:
[image]
> there is already a 5MB application in `/root` directory :)

Is this caused by compiling? Where can I change the parameters?

And, `free -m` show I have 186384 in total but I don't know the unit of it :(

luckfox-eng33, 2023-10-11:
Hello, there should be a problem with your image,
you can see board configuration file under the `/project/cfg/BoardConfig_IPC` folder，like[BoardConfig-SPI_NAND-NONE-RV1106_Luckfox_Pico_Max-IPC.mk](https:github.com/LuckfoxTECH/luckfox-pico/blob/main/project/cfg/BoardConfig_IPC/BoardConfig-SPI_NAND-NONE-RV1106_Luckfox_Pico_Max-IPC.mk),which contains partition configuration.
the unit returned by free is KB. 
If you want to check the memory information, you can use `cat /proc/meminfo`

bonuschild, 2023-10-11:
> Hello, there should be a problem with your image, you can see board configuration file under the `/project/cfg/BoardConfig_IPC` [...]

The image is with-board image which is not burn by myself. I will checkout this file, thx.

luckfox-eng33, 2023-11-16:
issues close

closed 2023-11-16 by luckfox-eng33