LuckfoxTECH/luckfox-pico #16: ./build.sh: line 1207: bc: command not found

versamodule, 2023-11-03:
Following the directions here: https://www.spotpear.com/index/study/detail/id/1113.html
When trying to build I get the following error. 

Which would you like? [0]: 0
[build.sh:info] switching to board: /home/jim/luckfox-pico-main/project/cfg/BoardConfig_IPC/BoardConfig-EMMC-NONE-RV1103_Luckfox_Pico-IPC.mk
**************************************
Check [OK]: dtc --version
**************************************
Check [OK]: makeinfo --version
**************************************
Check [OK]: gperf --version
**************************************
Check [OK]: dpkg --list |grep g++-.*-multilib
**************************************
Check [OK]: dpkg --list |grep gcc-.*-multilib
**************************************
Check [OK]: make -v
./build.sh: line 1207: bc: command not found
[build.sh:error] Running parse_partition_env failed!
[build.sh:error] exit code 127 from line 1207:
[build.sh:info]     part_size_bytes_b16="0x`echo "obase=16;$part_size_bytes"|bc`"
jim@DESKTOP-UUTBUQP:~/luckfox-pico-main$

chaguidiao, 2023-11-04:
sudo apt install bc

luckfox-eng33, 2023-11-16:
The dependency has been updated and was previously missing
* Installing Dependencies
```
sudo apt-get install -y git ssh make gcc gcc-multilib g++-multilib module-assistant expect g++ gawk texinfo libssl-dev bison flex fakeroot cmake unzip gperf autoconf device-tree-compiler libncurses5-dev pkg-config bc python-is-python3 passwd openssl openssh-server openssh-client vim file cpio rsync
```
If there is no problem, the issue will be close after one day

closed 2024-03-01 by andrethomas