LuckfoxTECH/luckfox-pico #20: build.sh fatal error when compiling GDB

set5una, 2023-11-17:
Fatal error is encountered when attempting to compile by invoking ./build.sh

```
source-cache.c:37:10: fatal error: srchilite/sourcehighlight.h: No such file or directory
 #include <srchilite/sourcehighlight.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [Makefile:1618: source-cache.o] Error 1
make[4]: *** Waiting for unfinished jobs....
In file included from /home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/vector:69,
                 from ./../gdbsupport/common-utils.h:24,
                 from ./../gdbsupport/common-defs.h:125,
                 from defs.h:28,
                 from solib-target.c:20:
/home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/bits/vector.tcc: In member function 'void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const long long unsigned int&}; _Tp = long long unsigned int; _Alloc = std::allocator<long long unsigned int>]':
/home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/bits/vector.tcc:413:7: note: parameter passing for argument of type 'std::vector<long long unsigned int>::iterator' {aka '__gnu_cxx::__normal_iterator<long long unsigned int*, std::vector<long long unsigned int> >'} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
In file included from /home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/vector:64,
                 from ./../gdbsupport/common-utils.h:24,
                 from ./../gdbsupport/common-defs.h:125,
                 from defs.h:28,
                 from solib-target.c:20:
/home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/bits/stl_vector.h: In member function 'void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = long long unsigned int; _Alloc = std::allocator<long long unsigned int>]':
/home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<long long unsigned int*, std::vector<long long unsigned int> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
In file included from /home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/vector:69,
                 from ./../gdbsupport/common-utils.h:24,
                 from ./../gdbsupport/common-defs.h:125,
                 from defs.h:28,
                 from remote.c:22:
/home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/bits/vector.tcc: In member function 'void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {static_tracepoint_marker}; _Tp = static_tracepoint_marker; _Alloc = std::allocator<static_tracepoint_marker>]':
/home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/bits/vector.tcc:413:7: note: parameter passing for argument of type 'std::vector<static_tracepoint_marker>::iterator' {aka '__gnu_cxx::__normal_iterator<static_tracepoint_marker*, std::vector<static_tracepoint_marker> >'} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
/home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/bits/vector.tcc: In member function 'virtual std::vector<static_tracepoint_marker> remote_target::static_tracepoint_markers_by_strid(const char*)':
/home/setsuna/Documents/RV1103/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/arm-rockchip830-linux-uclibcgnueabihf/include/c++/8.3.0/bits/vector.tcc:109:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<static_tracepoint_marker*, std::vector<static_tracepoint_marker> >' changed in GCC 7.1
    _M_realloc_insert(end(), std::forward<_Args>(__args)...);
    ^~~~~~~~~~~~~~~~~
make[3]: *** [Makefile:10066: all-gdb] Error 2
make[2]: *** [Makefile:854: all] Error 2
make[1]: *** [Makefile:29: all] Error 255
make[1]: Leaving directory '/home/setsuna/Documents/RV1103/luckfox-pico/sysdrv/tools/board/gdb'
make: *** [/home/setsuna/Documents/RV1103/luckfox-pico/sysdrv/tools/board/Makefile.tools.board.mk:32: board-build-gdb] Error 2
make: Leaving directory '/home/setsuna/Documents/RV1103/luckfox-pico/sysdrv'
[build.sh:error] Running build_sysdrv failed!
[build.sh:error] exit code 2 from line 518:
[build.sh:info]     make -C ${SDK_SYSDRV_DIR}
```

set5una, 2023-11-17:
Found a fix: append `--disable-source-highlight` to sysdrv/tools/board/gdb/Makefile as a `./configure` flag temporarily disables the missing module and allows compilation to continue. [Source](https://stackoverflow.com/a/68955819)

luckfox-eng33, 2023-11-20:
What is your OS?
Recommended OS is Ubuntu 22.04

set5una, 2023-11-20:
Archlinux, but afaik this should be fairly OS agnostic.

luckfox-eng33, 2023-11-21:
> Archlinux, but afaik this should be fairly OS agnostic.

Maybe you can try using the docker we provide to rule out the impact of the OS.

set5una, 2023-11-22:
> > Archlinux, but afaik this should be fairly OS agnostic.
> 
> Maybe you can try using the docker we provide to rule out the impact of the OS.

I failed to reproduce the issue with the Ubuntu docker image, however, I do believe adhering to upstream patches mentioned in #23 will improve compatibility across OSes, and I will continue to investigate the cause of this issue.

andrethomas, 2024-03-01:
Closing this one for now since it has an open PR

closed 2024-03-01 by andrethomas

itsmattkc, 2025-11-08:
This issue has managed to sneak its way back in. At some point, the fix in #23 was reverted and [sysdrv/tools/board/gdb/Makefile](https://github.com/LuckfoxTECH/luckfox-pico/blob/main/sysdrv/tools/board/gdb/Makefile) no longer has `--disable-source-highlight` in it.