LuckfoxTECH/luckfox-pico #21: Dropbear cannot use local user login in busybox

peterwillcn, 2023-11-17:
> assign a password for the root on the target board (this may require you to enable the passwd in busybox by make busybox-menuconfig in buildroot)"

Oh!! there's a whole new menu!!

For those who are following this trail, here is what I did to get dropbear working:

1, build dropbear and add it to your image/distribution

2, run "make busybox-menuconfig" and ensure that passwd is added to your image/distribution

3, build and reboot your board (making sure it picks up the new image).

4, set a root password

5, Run the following to generate your keys
```
dropbearkey -t ed25519 -f /etc/dropbear/dropbear_ed25519_host_key
dropbearkey -y -f /etc/dropbear/dropbear_ed25519_host_key | grep "^ssh-ed25519 " >> /etc/dropbear/authorized_keys
```
6, Start dropbear on port 22 
`/usr/sbin/dropbear -r /etc/dropbear/dropbear_ed25519_host_key -p 22`

7, Connect using ssh 
`ssh root@IP-address -p 22`

peterwillcn, 2023-11-17:
### issues: cannot use local user login in;

```
OpenSSH_9.4p1, LibreSSL 3.3.6
debug1: Reading configuration data /Users/alex/.ssh/config
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 21: include /etc/ssh/ssh_config.d/* matched no files
debug1: /etc/ssh/ssh_config line 54: Applying options for *
debug1: Authenticator provider $SSH_SK_PROVIDER did not resolve; disabling
debug1: Connecting to 172.20.0.169 [172.20.0.169] port 22.
debug1: Connection established.
debug1: identity file /Users/alex/.ssh/id_rsa type 0
debug1: identity file /Users/alex/.ssh/id_rsa-cert type -1
debug1: identity file /Users/alex/.ssh/id_ecdsa type -1
debug1: identity file /Users/alex/.ssh/id_ecdsa-cert type -1
debug1: identity file /Users/alex/.ssh/id_ecdsa_sk type -1
debug1: identity file /Users/alex/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /Users/alex/.ssh/id_ed25519 type 3
debug1: identity file /Users/alex/.ssh/id_ed25519-cert type -1
debug1: identity file /Users/alex/.ssh/id_ed25519_sk type -1
debug1: identity file /Users/alex/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /Users/alex/.ssh/id_xmss type -1
debug1: identity file /Users/alex/.ssh/id_xmss-cert type -1
debug1: identity file /Users/alex/.ssh/id_dsa type -1
debug1: identity file /Users/alex/.ssh/id_dsa-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_9.4
debug1: Remote protocol version 2.0, remote software version dropbear_2022.83
debug1: compat_banner: no match: dropbear_2022.83
debug1: Authenticating to 172.20.0.169:22 as 'root'
debug1: load_hostkeys: fopen /Users/alex/.ssh/known_hosts2: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts2: No such file or directory
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ssh-ed25519
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: SSH2_MSG_KEX_ECDH_REPLY received
debug1: Server host key: ssh-ed25519 SHA256:ACqtn2Z6CfyGZhW8/p7SinxshGoGebvbuV+GS7c/LeI
debug1: load_hostkeys: fopen /Users/alex/.ssh/known_hosts2: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts2: No such file or directory
debug1: Host '172.20.0.169' is known and matches the ED25519 host key.
debug1: Found key in /Users/alex/.ssh/known_hosts:25
debug1: rekey out after 134217728 blocks
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey in after 134217728 blocks
debug1: get_agent_identities: bound agent to hostkey
debug1: get_agent_identities: ssh_fetch_identitylist: agent contains no identities
debug1: Will attempt key: /Users/alex/.ssh/id_rsa RSA SHA256:NOLRrCTGoduC3IdY9Ip0Hnj/yZSs1vH5ZOd2BIxtWeI
debug1: Will attempt key: /Users/alex/.ssh/id_ecdsa
debug1: Will attempt key: /Users/alex/.ssh/id_ecdsa_sk
debug1: Will attempt key: /Users/alex/.ssh/id_ed25519 ED25519 SHA256:duzik4W09Q3dWrmJX6sIuCD4B6FBG0gKH4uetT3DQIU
debug1: Will attempt key: /Users/alex/.ssh/id_ed25519_sk
debug1: Will attempt key: /Users/alex/.ssh/id_xmss
debug1: Will attempt key: /Users/alex/.ssh/id_dsa
debug1: SSH2_MSG_EXT_INFO received
debug1: kex_input_ext_info: server-sig-algs=<ssh-ed25519,sk-ssh-ed25519@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,sk-ecdsa-sha2-nistp256@openssh.com,rsa-sha2-256,ssh-rsa,ssh-dss>
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey,password
debug1: Next authentication method: publickey
debug1: Offering public key: /Users/alex/.ssh/id_rsa RSA SHA256:NOLRrCTGoduC3IdY9Ip0Hnj/yZSs1vH5ZOd2BIxtWeI
debug1: Authentications that can continue: publickey,password
debug1: Trying private key: /Users/alex/.ssh/id_ecdsa
debug1: Trying private key: /Users/alex/.ssh/id_ecdsa_sk
debug1: Offering public key: /Users/alex/.ssh/id_ed25519 ED25519 SHA256:duzik4W09Q3dWrmJX6sIuCD4B6FBG0gKH4uetT3DQIU
debug1: Authentications that can continue: publickey,password
debug1: Trying private key: /Users/alex/.ssh/id_ed25519_sk
debug1: Trying private key: /Users/alex/.ssh/id_xmss
debug1: Trying private key: /Users/alex/.ssh/id_dsa
debug1: Next authentication method: password
root@172.20.0.169's password:
debug1: Authentications that can continue: publickey,password
Permission denied, please try again.
root@172.20.0.169's password:
debug1: Authentications that can continue: publickey,password
Permission denied, please try again.
root@172.20.0.169's password:
```

luckfox-eng33, 2023-11-21:
> Oh!! there's a whole new menu!!
>
> For those who are following this trail, here is what I did to get dropbear working:
>
> 1, build dropbear and add it to your image/distribution
>
> 2, run "make [...]

Hello, I don’t quite understand your question.
Are you using the latest SDK?
SSH is enabled by default in the latest SDK, and you can log in normally.

andrethomas, 2024-03-01:
Solution provided in https://github.com/LuckfoxTECH/luckfox-pico/issues/21#issuecomment-1816450974

closed 2024-03-01 by andrethomas