LuckfoxTECH/luckfox-pico #22: I2S support

casterplay, 2023-11-18:
Hi, does LUCKFOX Pico Pro or Max support I2S protocol. Simply, can I talk to a audio ADC or DAC (like PCM5102) with my Luckfox Max?

SantiagoSotoC, 2024-01-06:
I am also interested in this

andrethomas, 2024-02-10:
I'm also interested in this

luckfox-eng33, 2024-03-06:
RV1103/6 has MIC and LINEOUT interfaces, which seems to be a better solution.
In the next version of the hardware, we will provide microphone and speaker interfaces, so stay tuned

casterplay, 2024-03-06:
Sorry but MIC and LINEOUT is not a good solution, maybe for home tinkering... a I2S interface is what's needed so I can use whatever DAC or ADC I need, two I2S channels is even better...  

ESP32, RP20, Teensy and more already have I2S...

andrethomas, 2024-03-07:
The RV1106 silicon definitively has I2S support, according to datasheet

[image]

In the RV1106 datasheet there are pins listed with I2S functionality as follows

```
106 UART0_CTS_M1/I2S0_SDO1_SDI3/VO_LCDC_D15/PWM4_M1/I2C3_SDA_M0/PRELIGHT_TRIG_OUT/GPIO2_A7
107 UART0_RTS_M1/I2S0_SDO2_SDI2/VO_LCDC_D14/PWM2_M1/I2C3_SCL_M0/FLASH_TRIG_OUT/GPIO2_A6_d
109 SDMMC1_D1_M0/I2S0_SCLK/VO_LCDC_D8/UART1_CTS_M1/I2C4_SDA_M0/GPIO2_A0_d 
110 SDMMC1_D1_M0/I2S0_SCLK/VO_LCDC_D8/UART1_CTS_M1/I2C4_SDA_M0/GPIO2_A0_d
111 SDMMC1_CLK_M0/I2S0_MCLK/VO_LCDC_D10/GPIO2_A2_d 
112 SDMMC1_CMD_M0/I2S0_SDO3_SDI1/VO_LCDC_D11/GPIO2_A3_
113 SDMMC1_D3_M0/I2S0_SDO0/VO_LCDC_D12/UART1_TX_M1/GPIO2_A4_d 
114 SDMMC1_D2_M0/I2S0_SDI0/VO_LCDC_D13/UART1_RX_M1/GPIO2_A5_d
```

I do not know for sure how those pin numbers from the datasheet map to the broken out pins on the wiki found here:

[https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPIO](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPIO)

BUT, although I cannot get the math suggested in the wiki for pin numbers to work well for comparing the two set of pin numbers there are some instances where the GPIOxxx name of the pin on the Wiki corresponds with the GPIOxxx name of the pin numbers listed for I2S in the datasheet.

If someone else has any clue as to how to confirm the pin mappings then that may be helpful.

Otherwise its going to be a case of taking an I2S codec (I think I have some PCM3060's in a tray) and connecting it to the board and see what happens - probably it will need some tweaking in the kernel and then I guess there may or may not be support for the I2S silicon from the kernel source provided by rockchip.

andrethomas, 2024-03-11:
Case in point for the wiki vs. datasheet pin numbering...

In the datasheet the RV1106, for example, has only 128 pins... but the wiki uses pin numbers as high as 145? How do the pin numbering on the board correlate with the physical pins on the processor?

[image]

[image]

andrethomas, 2024-03-11:
So according to the board pinout above, read with the pin names/descriptions, the most likely candidate for I2S access is likely:

```
21: I2S0_SDO0 (GPIO2_A4_d)
22: I2S0_SDI0 (GPIO2_A5_d)
24: I2S0_SCLK (GPIO2_A0_d)
26: I2S0_MCLK (GPIO2_A2_d)
```

andrethomas, 2024-03-11:
The pins are different for the different boards, but for the image above which is the board I have being a Pro Max, the most likely candidate at this point for I2S is the pins at the bottom right in the board pinout image:

```
21: I2S0_SDO0 (GPIO2_A4_d)
22: I2S0_SDI0 (GPIO2_A5_d)
24: I2S0_SCLK (GPIO2_A0_d)
26: I2S0_MCLK (GPIO2_A2_d)
```

fearpro13, 2024-09-08:
> RV1103/6 has MIC and LINEOUT interfaces, which seems to be a better solution. In the next version of the hardware, we will provide microphone and speaker interfaces, so stay tuned

luckfox pico (rv1103) audio codec in/out pins connected to board pads(bottom side)
luckfox pico mini B (rv1103) has audio codecs pins too, but they aren't connected to anywhere

Is there any possible solution to get audio from chip on pico-mini A/B ?
Currently using one of pwm(pin 10) as audio output - terrible and consumes all cpu

Btw, is there any possibility to run irq-based pwm control code? Same as stm32 ones? 

Thanks!
[image]

VincentBruinink, 2025-05-03:
Orange PI, Banana PI, LUCKFOX, all companies that don't want you to use I2S. All they provide is crappy internal audio codecs with very poor S/N.