LuckfoxTECH/luckfox-pico #28: No DHCP address on Pico Mini B

padcom, 2023-11-26:
I just bought 2 Luckfox Pico Mini B boards and was trying to test it out on Linux (Mint 21.2).

With its bigger brother I was able to just burn the image onto an SD card and it would provide an IP address to my computer with no problems. With the Mini B the docs say there is already an image burned onto the on-board flash.

I tried updating it with the latest available at https://drive.google.com/drive/folders/1sFUWjYpDDisf92q9EwP1Ia7lHgp9PaFS?usp=drive_link using the linux version of (update_tool)[https://files.luckfox.com/wiki/Core3566/upgrade_tool_v2.17.zip] and the update went all ok.

However, I still don't get an IP address assigned from the little board on `usb0`.

Is DHCP enabled on that little board?

luckfox-eng33, 2023-11-27:
https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Login-Network
Default IP address is 172.32.0.93

padcom, 2023-11-27:
[image]

Would it kill you to have a section on how to connect over SSH? Or for that matte "Default network configuration section"?

If someone is looking for the default IP address checking the ADB section is the last place to look for it...

luckfox-eng33, 2023-12-01:
I have informed my colleagues and they will update it later