LuckfoxTECH/luckfox-pico #31: LuckFox Pico Pro Max: which converter cable for an Arducam 16MP?

brainstorm, 2023-12-06:
I was looking at the [Luckfox Pico Pro Max schematic](https://wiki.luckfox.com/Luckfox-Pico/Datasheets) and I noticed an unusual 20P camera connector:

[image]

[Normally it's either 15-pin or 22-pin](https://www.arducam.com/raspberry-pi-camera/connector-type-pinout/) and there are convenient cables to adjust pitch and smaller boards:

[image]

I.e, Raspberry Pi Zero 2W has a 22 pin connector like so:

[image]

Are you planning to sell those custom FFC 0.5mm 20P adapter cables for other cameras other than the "SC3336 3MP Camera Module (B)" which uses [this one](https://item.szlcsc.com/7901783.html) that seems to be incompatible with i.e [any of those Arducam cameras](https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/16MP-IMX519/)?

I'm thinking on manufacturing one myself but before I do that, I wanted to check w/ you... @luckfox-eng33 ?

gehee, 2024-01-08:
It would also be very nice to have a pico MAX "plus" with 4 mipi lanes, instead of two.

Would the arducam 16MP sensor work with two lanes only?

luckfox-eng33, 2024-01-09:
RV1103 supports a maximum of 4M@30fps 
RV1106 supports a maximum of 5M@30fps
In addition, other cameras may need to be debugged by themselves.

gehee, 2024-01-09:
If it can do 5M@30fps, can it do 2M@60fps ? It's less pixel/s.
It would be nice to have a camera with higher framerate.

luckfox-eng33, 2024-01-09:
Theoretically it's possible, but it hasn't been tested

gehee, 2024-01-09:
Would it be possible to make a pico compatible with raspberry PI cameras? and with a faster ISP?
Its size makes it a perfect fit for a HD system broadcasting  OTA from a drone.

luckfox-eng33, 2024-01-09:
If you need low latency, you may need to write a streaming program, but the RKIPC latency may exceed 1.5S

platima, 2024-05-04:
Hey unfortunately it's not going to be easily possible.

I mapped the pins between different connectors at https://plati.ma/wp-content/uploads/2024/04/Schematic_MIPI-CSI_2024-04-28.pdf and then drafted a list of what has what at https://github.com/platima/sbc-cameras/blob/main/readme.md

Luckfox is annoyingly unique in these messed up 22 pin MIPI CSI connectors -_- I am considering fabricating an adapter board for it though if anytone is interested? Would be a few bucks, depending on the quantity.

platima, 2024-05-04:
> If it can do 5M@30fps, can it do 2M@60fps ? It's less pixel/s. It would be nice to have a camera with higher framerate.

Should work - it's just about the memory reserved (64MB by default in RV1106 I think) and the bitrate. 5M at 30fps is around 2.4Gbps (yuv422) which is about the max of MIPI CSI2 using 4-lanes from what I know. As long as you don't exceed this, and the ISP can handle it, then you should be fine.

platima, 2024-05-04:
> It would also be very nice to have a pico MAX "plus" with 4 mipi lanes, instead of two.
> 
> Would the arducam 16MP sensor work with two lanes only?

Possibly, as it usually negotiates, but you'd be capped at about ~9fps if it does work.