LuckfoxTECH/luckfox-pico #32: LuckFox Pico MAX  -->  SD CARD BOOT FAIL

Francky400, 2023-12-08:
Hello,

I try to boot LuckFox Pico MAX on SD Card (Not on SPI NAND FLASH)
I have erased SPI NAND FLASH before, lik section "Erasing SPI NAND Flash" 

I have following the section "LuckFox Pico Pro/Max image burning", "SD Card Flashing" of your wiki web "https://wiki.luckfox.com/Luckfox-Pico"
The flash process seems to be OK (no error messages)

But, when i plug the SD Card, into slot of LuckFox pico MAX and turn on module, there is nothing, no led flashing, no UART2 RS232 traces, no ADB connexion.

If i try to put SD card in a SD card reader, no partition table is visible.

1) to boot on SD Card, is there a specific firmware image for LuckFox pico MAX ?

2) If yes, where i can download the firmware for flash SD card for the LuckFox pico MAX please ?

3) Are there any specific options to put in the file "env.txt" to enable serial debugging on the UART2 ?

Thanks in advance
Best Regards

HANSEN Franck

luckfox-eng33, 2023-12-08:
Hello
The SD card firmware is different from the NAND FLASH firmware . 
The boot media and file formats of the two are different.

It is recommended that you use the SDK to recompile the corresponding firmware . 
You need to modify the MK file you use. 
Refer to the MK file of Luckfox_Pico to modify the corresponding **RK_BOOT_MEDIUM** and **RK_UBOOT_DEFCONFIG_FRAGMENT** and recompile.

github can't provide a firmware because it's too big

You seem to contact our agent, because it is the same problem, you should have obtained the corresponding mirror

So we don't provide a mirror image here

Francky400, 2023-12-08:
Hello, ok for that, i will try to recompile SDK with the good options

Please, how can enable serial debugging for LuckFox pico MAX when boot on SD Card ?

Regards

luckfox-eng33, 2023-12-08:
> Hello, ok for that, i will try to recompile SDK with the good options
> 
> Please, how can enable serial debugging for LuckFox pico MAX when boot on SD Card ?
> 
> Regards

It's all enable by default, don't worry

Francky400, 2023-12-08:
Hi,

I have try your image : https://cdn.static.spotpear.com/uploads/download/diver/luckfox-pico-promax/img/Luckfox-Pico-Max-SD.7z
I have burn it on SD Card with your flashing tool --> SD Card flash ok, no errors

I have put this SD Card on a LuckFox Pico MAX module
I have put a serial to USB module on pins 1, 2 and 3 on the LuckFox Pico Max module (like "Serial Debugging" section of your Wiki")

The module boot from SD Card, i can connect to it with ADB, but i never get debug traces on serial

I use minicom under a ubuntu PC, to monitor serial debug, but i have nothing (configuration of COM : 115200 8N1 no flow control )
I use command : sudo minicom -D /dev/ttyUSB0 
With other boards, like Raspberry Pi, i get serial debug traces in this configuration

Please can you check this

Best Regards

HANSEN Franck

----- Mail original -----

luckfox-eng33, 2023-12-08:
What USB TO UART module are you using? Make sure the communication logic level is 3.3V
Maybe you can look at the information output by `lsusb`

Francky400, 2023-12-08:
Hi,

lsusb: 

Bus 001 Device 087: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port

I try other : 

Bus 001 Device 099: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

I never have serial debug traces, this boh USB to serial converter are full fonctionally with 3.3v levels

Regards

HANSEN Franck

----- Mail original -----

closed 2024-03-01 by andrethomas