LuckfoxTECH/luckfox-pico #36: RV1106 Complete User Manual data sheet

Francky400, 2023-12-10:
HI,

Please, where i can download the full complete Rockchip RV1106 Datasheet (user reference manual, registers description, .....)

Thanks in advance

HANSEN Franck

luckfox-eng33, 2023-12-11:
Hello
Due to problems with ROCKCHIP, we cannot provide the corresponding files directly. You can check doc(https://files.luckfox.com/wiki/Luckfox-Pico/PDF/doc.7z） first

EricH45, 2023-12-23:
Hello,
The link to the docs is not usable. Does it changed ?
Regards,
Eric H

luckfox-eng33, 2023-12-23:
> Hello, The link to the docs is not usable. Does it changed ? Regards, Eric H

It may be a problem with github. You can download it from our wiki.
https://wiki.luckfox.com/Luckfox-Pico/Datasheets
[image]

closed 2024-03-01 by andrethomas