LuckfoxTECH/luckfox-pico #39: LuckFox MAX --> Uboot customization

Francky400, 2023-12-11:
Hi,

I try to add a boot delay for few seconds, to have time to hit a key (under rs232 debug console), to stop autoboot and enter in u-boot monitor, for modify some environment variables.

This i the behaviour by default in -boot, i have many other boards with u-boot work with this delay.

Unfortnatelly, i have modify the uboot  defconfig file "rv1106_defconfig" to change "BOOTDELAY" from 0 to 4, recompile uboot, update uboot.img to SDCARD with your flashing tool.

I never see the delay to interrupt autoboot and enter in the uboot monitor.
I cannot interrupt autoboot, the boot scripts are running always

Please, can you tell me the procedure to enable possibility to stop autoboot when hit a key please ?

My serial debug console is OK, when the boot is finish, i can enter login and password to enter into linux

Best Regards
HANSEN Franck

luckfox-eng33, 2023-12-12:
Hello
Modifying `CONFIG_BOOTDELAY` should be useful. You can try `build.sh clean` first and then compile `build.sh`
You can also press CTRL+C crazily after booting to enter uboot

Francky400, 2023-12-15:
Hello,

Perfect, thanks a lot

Franck

closed 2024-03-01 by andrethomas