LuckfoxTECH/luckfox-pico #42: Luckfox Max --> take photos or videos with the camera and save them on SPI NAND FLASH or SD CARD

Francky400, 2023-12-15:
Hello,

By default, the rkipc application is started at linux startup, providing an RTSP stream through the network.

My need is not to generate a video stream.
I would like to know how to take photos or videos with the camera connected to the lucfox (SC3336) and record these photos or videos on the SPI NAND FLASH or the SD Card of the Luxfox module.

Certainly, do not launch the application rkipc, to free the resource "camera", but what command or application to use to make photos or videos?

Thanks in advance
HANSEN Franck

luckfox-eng33, 2023-12-15:
1. Run the image
2. you have to kill rkipc by using the command 'killall rkipc'
3. You can get some 'raw pictures'（look no good） through some simple command ,
* Use `simple_venc_jpeg -I 0-w 1920-h 1080 ` and a few carriage returns, you'll get some jpg in /tmp
* Use `sample_vi -w 1920-h 1080 -a /etc/iqfiles/-l 10-o /userdata/`, you will get a vi_0.bin file in /userdata, which is a yuv file. Requires special software (such as yuvplaer) to open
4. Also can see reference [opencv-mobile](https://github.com/nihui/opencv-mobile) ，The author has introduced how luckfox-pico is used on [Zhihu](https://zhuanlan.zhihu.com/p/670191385) (like quora)

luckfox-eng33, 2023-12-15:
> 4. Also can see reference [opencv-mobile](https://github.com/nihui/opencv-mobile) ，The author has introduced how luckfox-pico is used on [Zhihu](https://zhuanlan.zhihu.com/p/670191385) (like quora)

And Our [wiki](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-opencv-mobile) provides a more detailed way to use it
https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-opencv-mobile

Francky400, 2023-12-15:
Thanks a lot,

i will try following this tutorial
Franck

closed 2024-03-01 by andrethomas

EduardoArtioli, 2025-04-22:
@Francky400 

Did you manage to implement photo and video recording? If you have an example of how to do it, it would help a lot.