LuckfoxTECH/luckfox-pico #43: Configuration Interface for Luckfox Pico Mini A

versamodule, 2023-12-15:
I have the Pico Mini A, and I would like to add Ethernet to it. 
I have bought the same RJ45 ethernet connector that is used on the LuckFox Pico Plus. 
From what I gather it need to re-compile the kernel to add Ethernet. I was reading here: 
https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Menuconfig/

that is says:

> Execute the following command to open the configuration interface:
> 
> cd <SDK directory>/sysdrv/source/kernel
> cp ./arch/arm/configs/luckfox_rv1106_linux_defconfig .config
> make ARCH=arm menuconfig
> 

But this is not for my pico Mini A, so what would I do here ?

closed 2023-12-16 by versamodule

luckfox-eng33, 2023-12-18:
Refer to the mk file, it seems you have found the solution.

platima, 2024-06-02:
Hey @versamodule 

@luckfox-eng33 is trying to "teach a man to fish" :P

Not wrong though, this has the exact instruction: https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-SDK#4-modifying-device-tree

I have also just built a Luckfox Pico Mini (B) Ubuntu image with gmac enabled. Works on Mini A too: https://github.com/platima/sbc-images/tree/main/Luckfox/Pico

Enjoy!