LuckfoxTECH/luckfox-pico #49: ADB and RNDIS not working.

Pengunix, 2023-12-19:
Device : luckfox pico
Official SDK with default config.
and i got this error while booting.

```
/etc/init.d/S50usbdevice: line 144: can't open : no such file
Debug: configfs_init
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip': No such file[    3.550916] rockit: Unknown symbol rkisp_rockit_get_ or directoryispdev (err -2)
[    3.551136] rockit: Unk
nown symb/etc/init.d/S50usbdeviceol rkisp: line 76: can't create /sys/kernel/conf_roig/usb_gadget/rockchip/idVendorinsmod: can't insert 'rockit.ko': unknown symbol in r
: nonexistent directory
ckit_free_tb_s/etc/init.d/S50utream_buf sbdevice(e: line rr77:  -can't create /sys/kernel/conf2)ig/usb_gadget/rockchip/bcdDevice: nonexistent directory
/etc/init.d/S50usbdevice: line 78: can't create 
[   /sys/kernel/config/usb_gadget/roc 3.551179] rockit: kchip/bcdUSB/
ctorywn symbol rkis: nonexistent direp_rockit_function_register (err -2)
     [    
3.551265] rockit: Unknown symbol rkiinsmod_ko.sh: line 78: /oem/usr/ko/insmsp_rockit_od_wifi.shresum: not founde_str
eam (err -2)
[    3.551312] rockit: Unknown symbol rkisp_rockit_config_stream (err -2)
[    3.551383] rockit: Unknown symbol rkisp_rockit_get_tb_stream_info (err -2)
[    3.551466] rockit: Unknown symbol rkisp_rockit_pausln: /usr/www/usee_stream (erdata: No such firr -2)
[    le or directory3.557409] rockit: Unknown symbol rkisp_rockit_get_
ispdev (err -2)
[    3.557601] rockit: Unknown symbol rkisp_rockit_free_tb_stream_buf (err mkdir: can't cre-2)
[    3.557645] rockiate directory '/st: Unknown ys/kernel/config/symbol rkisp_rockit_function_register (err -2)
[    3usb_gadget/rockch.5580ip/strings/0x409'12] rockit: Unknown symbol rkisp_rocki: No such file ort_resum directory
e_stream (err -2)
ln: /usr/www/usbockit: Unknown symbol rkisp_rockit_config_stream (err -2)
                [    3.5581470: No such file o] rockit: Unknown symbol rkisp_rockit_get_tb_stream_info (err directoryr -2)
[  
  3.558484] rockit: Unknown symbol rkisp_rockit_pause_stream (err -2)
eth HWaddr cfg ok
ln: /usr/www/sdcard: No such file or directory
/etc/init.d/S50usbdevice: line 84: can't create /sys/kernel/config/usb_gadget/rockchip/strings/0x409/serialnumber: nonexistent directory
/etc/init.d/S50usbdevice: line 85: can't create /sys/kernel/config/usb_gadget/rockchip/strings/0x409/manufacturer: nonexistent directory
/etc/init.d/S50usbdevice: line 86: can't create /sys/kernel/config/usb_gadget/rockchip/strings/0x409/product: nonexistent directory
ifconfig: SIOCGIFFLAGS: No such device
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/uac1.gs0': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/uac2.gs0': No such file or directory
md5sum: can't open '/tmp/rkipc-factory-config.ini': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/ffs.adb': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/ffs.ntb': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/mtp.gs0': No such file or directory
Error: not found rkipc.ini !!!
/oem/usr/bin/RkLunch.sh: exit: line 110: Illegal number: -1
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/rndis.gs0': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/acm.gs6': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/mass_storage.0': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/uvc.gs6': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/functions/hid.usb0': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/configs/b.1': No such file or directory
mkdir: can't create directory '/sys/kernel/config/usb_gadget/rockchip/configs/b.1/strings/0x409': No such file or directory
/etc/init.d/S50usbdevice: line 93: can't create /sys/kernel/config/usb_gadget/rockchip/os_desc/b_vendor_code: nonexistent directory
/etc/init.d/S50usbdevice: line 94: can't create /sys/kernel/config/usb_gadget/rockchip/os_desc/qw_sign: nonexistent directory
/etc/init.d/S50usbdevice: line 95: can't create /sys/kernel/config/usb_gadget/rockchip/configs/b.1/MaxPower: nonexistent directory
ln: /sys/kernel/config/usb_gadget/rockchip/os_desc/b.1: No such file or directory
/etc/init.d/S50usbdevice: line 420: can't create /sys/kernel/config/usb_gadget/rockchip/idProduct: nonexistent directory
ln: /sys/kernel/config/usb_gadget/rockchip/configs/b.1/f1: No such file or directory
ln: /sys/kernel/config/usb_gadget/rockchip/configs/b.1/f2: No such file or directory
/etc/init.d/S50usbdevice: line 368: can't create /sys/kernel/config/usb_gadget/rockchip/configs/b.1/strings/0x409/configuration: nonexistent directory
mount: mounting adb on /dev/usb-ffs/adb failed: No such device
ls: /sys/class/udc/: No such file or directory
/etc/init.d/S50usbdevice: line 425: can't create /sys/kernel/config/usb_gadget/rockchip/UDC: nonexistent directory
ifconfig: SIOCSIFADDR: No such device
ifconfig: SIOCGIFFLAGS: No such device
ifconfig: usb0: error fetching interface information: Device not found
current_ip = 
TARGET_IP = 172.32.0.93
luckfox : set usb0 ip
ifconfig: SIOCSIFADDR: No such device
ifconfig: usb0: error fetching interface information: Device not found

luckfox : set usb0 ip
ifconfig: SIOCSIFADDR: No such device
ifconfig: usb0: error fetching interface information: Device not found

luckfox : set usb0 ip
ifconfig: SIOCSIFADDR: No such device
ifconfig: usb0: error fetching interface information: Device not found

luckfox : set usb0 ip
ifconfig: SIOCSIFADDR: No such device
ifconfig: usb0: error fetching interface information: Device not found

luckfox : set usb0 ip
ifconfig: SIOCSIFADDR: No such device
ifconfig: usb0: error fetching interface information: Device not found

luckfox : set usb0 ip
ifconfig: SIOCSIFADDR: No such device
ifconfig: usb0: error fetching interface information: Device not found

luckfox : set usb0 ip
ifconfig: SIOCSIFADDR: No such device
ifconfig: usb0: error fetching interface information: Device not found

luckfox : set usb0 ip
ifconfig: SIOCSIFADDR: No such device
ifconfig: usb0: error fetching interface information: Device not found
```

luckfox-eng33, 2023-12-19:
It seems that the file system and files are missing. 
If possible, please pull the SDK again and compile it, or use the image we provide first.

Pengunix, 2023-12-20:
I redownloaded the sdk, and it ... works, i dont know whats wrong...

closed 2023-12-20 by Pengunix