LuckfoxTECH/luckfox-pico #51: Luckfox Pico MAX --> NEED Toolchain based on glic (not uclibc)

Francky400, 2023-12-21:
Hi,

The default toolchain in SDK for luckfox pico MAX (RV1106), is based on crosstool-ng uclibc

Unfortunately, i need, for some components that i must add in the buildroot rootfs, a toolchain based on glibc 
I know that the glibc libary is largest in terms of space, but i will work with micro SD card and not on SPI NAND FLASH

I have made some tests, 
i have generate a new toolchain with crosstool-NG, based on glibc, the generated toolchain is ok
I have add, "runtime_lib" directory with file "lib.tar.bz2" with all library files inside.

I have modify, "BoardConfig-EMMC-Buildroot-RV1106_Luckfox_Pico_Pro_Max-IPC.mk" to select the new toolchain based on glibc
I have modify buildroot configuration, to select the new toolchain based on glibc, i have select glibc in buildroot, i have add all packages that i need.

When i run : ./build.sh
 
u-boot build OK
kernel build OK
rootfs build OK

when the build process is up to  "media", there are a lot of errors and build process STOP
When i take a look in some directories, for example "/SDK/luckfox-pico/media/rga" , there are only precompiled libraries for : 

release_rga_rk3588_aarch64-rockchip1031-linux-gnu
release_rga_rv1106_arm-rockchip830-linux-uclibcgnueabihf
release_rga_rv1126_arm-rockchip830-linux-gnueabihf

Same problem in directories : "SDK/luckfox-pico/media/isp", "SDK/luckfox-pico/media/mpp", ....

Please, can you work on the SDK to add option for select toolchain based on glibc for RV1106, for the entire generation process works with a glibc based toolchain.

Best Regards

HANSEN Franck

luckfox-eng33, 2023-12-22:
Hello Franck，
This is a good proposal, but it is a pity that Rockchip does not provide us with the corresponding SDK. 
If we make modifications, the workload will be relatively large and we will face many uncertainties.
We will contact rockchip and maybe we will add this option in the future, but it is expected to take a long time.

Francky400, 2023-12-22:
Hello,

I think the glibc support is a real added value for the luckfox Pico module.
In image processing applications using NPU support, and using the possible customization of the root file system using buildroot, it is absolutely necessary to add some packages that require glibc support.

I work in the aerospace industry in France, and we would like to use the Luckfox Pico MAX module, for a new monitoring application based on image processing. We absolutely need glibc support.

The average annual volume is about 2000 pieces if the prototype based on this module gives satidfaction.

We would like to know, before embarking on a study phase on this project, if Rockchip is ready to add to its SDK, for the RV1106, the suppport of the glibC

Best Regards
HANSEN Franck

ludiazv, 2023-12-30:
+1
Having glibc as an option would be great plus for these boards.

@Francky400 AFAIK build root here is only used to generate the rootfs so you can use the glibc toolchain to compile buildroot only using the provided toolchain for the rest. 

I think changing this lines with a different ```CROSS_COMPILE``` will do de trick:

https://github.com/LuckfoxTECH/luckfox-pico/blob/43b945a541a37c6cdcb157518ead3c5022150add/sysdrv/Makefile#L529C1-L531C1

The downside is that you can't use the software include in media,oem or userdata as this userland binaries are complied against uClibc. This could be overcome adding ld and uClibc .so to the root fs but not sure if there is any collision between glibc so names and uClibc. This way both libcs could coexist.

Best,

Francky400, 2023-12-31:
Hi,

Thanks for your message.
We have already test, in a previous application glibC and uClibc in the same target, but unfortunately that not work, there are library collisions.

I know that the rootfs is building with no problems with glibc, we have change toolchain variable, and build process is OKAY for uboot, kernel and rootfs.
The problem is that the build process stop when the "media" package is try to build.
We dont get images files for SD Card.

Have you request to Rockchip, the way to add toolchain with glibc for Luckfox Pico MAX (RV1106G3) in SDK please ?
The best way is to obtain a SDK with a selectiion option with glibC

Best Regards
HANSEN Franck

ludiazv, 2023-12-31:
I see, Im not using the camera in my application so I'm simply not including media in my image. If you need the camera I think most of the rockchip utils for the camera are located there. Rocchip should provide versions of those libraries in glibc and musl in my opinion.

Regarding glibc and ulibC side by side in the same system I have not previous experience but I managed in the past to have musl & glibc solving the collisions with LD_PRELOAD_PATH. If your set of dependencies are known and controlled (as it typically in the case of embedded applications) you can try this route at least to test if the boards works for at the functional and performance level. 

Happy new year,

Francky400, 2024-01-01:
Hi,

We need camera for made control, we have choose this module Luckfox Pico MAX because this is a feature documented in the SDK.
We would like to know if Rockchip can modify its SDK for the RV1106, to add a choice option to work with the glibC.
We know that there is a little work to recompile the libraries contained in the media project, oem and userdata.

Please take a request to Rockchip for that.

Happy New Year to all Users and Luckfox team

HANSEN Franck

luckfox-eng33, 2024-01-03:
Hello,
We try to communicate with rockchip. 
But unfortunately rockchip told us that they do not intend to support rv1106 to support glibc.
Happy New Year, everyone.

luckfox-eng33, 2024-01-05:
hi
The doc provides basic GPIO-related registers. 
[Rockchip_RV1103_User_Manual_GPIO.pdf](https://github.com/LuckfoxTECH/luckfox-pico/files/13838396/Rockchip_RV1103_User_Manual_GPIO.pdf)
[Rockchip_RV1106_User_Manual_GPIO.pdf](https://github.com/LuckfoxTECH/luckfox-pico/files/13838397/Rockchip_RV1106_User_Manual_GPIO.pdf)

If you need more register information, you may need to sign  NDA. 
Please contact us privately for details.

Francky400, 2024-01-08:
Hi,

Thanks for this documentation, this is a good starting point.
I would like to get the full documentation, to have explanations of all the features, such as the interruption mechanism and the registers involved

I can sign NDA
Following, is my professional emai, you can send to this email (with copy to my personnal email) the process and documents to be signed.

My professional email:
***@***.***

My personnal email:
***@***.***

I am waiting for documents to be signed

Best Regards

HANSEN Franck
 

----- Mail original -----

luckfox-eng33, 2024-01-11:
> Hi, Thanks for this documentation, this is a good starting point. I would like to get the full documentation, to have explanations of all the features, such as the interruption mechanism and the [...]

Hello
Did you know your account is blocked and I can't see the exact email?

Francky400, 2024-01-11:
Hello,

No i dont know that my account is blocked, it was a mistake, not a intentionnal message.
Can you reactivate my account, please. 

Following my previous message:
--------------------------------------------------------------------------------------------
Hi, 
Thanks for this documentation, this is a good starting point. 
I would like to get the full documentation, to have explanations of all the features, such as the interruption mechanism and the registers involved.

I can sign NDA Following, is my professional emai, you can send to this email (with copy to my personnal email) the process and documents to be signed. 

My professional email: ***@***.***
My personnal email: ***@***.*** 

I am waiting for documents to be signed 
Best Regards 

HANSEN Franck 
------------------------------------------------------------------------------------------

----- Mail original -----

luckfox-eng33, 2024-01-11:
Maybe you can check my homepage. My email is on my homepage. I look forward to your email.

closed 2024-03-01 by andrethomas

vectorcrumb, 2024-03-06:
> Hello, We try to communicate with rockchip. But unfortunately rockchip told us that they do not intend to make rv1106 to support glibc. Happy New Year, everyone.

@luckfox-eng33 Has rockchip indicated any possibility of supporting glibc or musl by now or should we consider this answer as final? I'm trying to run ROS in a docker container, but docker-engine requires glibc/musl to get packaged in buildroot.

pengphei, 2024-05-28:
I have made rv1106 & rv1103 running in openwrt(glibc based), also camera worked, but the isp (rkaiq_3A_server) not work, it will run **segment** when lunching:

Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/rkaiq_3A_server...

warning: exec file is newer than core file.
[New LWP 2661]
[New LWP 2660]
[New LWP 2662]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
Core was generated by `/usr/bin/rkaiq_3A_server'.
Program terminated with signal SIGSEGV, Segmentation fault.
--Type <RET> for more, q to quit, c to continue without paging--
#0  0xa6b63208 in memcpy () from /lib/libc.so.6
[Current thread is 1 (Thread 0xa6ab53c0 (LWP 2661))]
(gdb) bt
#0  0xa6b63208 in memcpy () from /lib/libc.so.6
#1  0xa6edc414 in AecCalibConfig (pCalib=0xa61174f0, pAecCfg=0xa611c460)
    at ../../../../algos/ae/rk_aiq_ae_algo.cpp:843
#2  0xa611d9e0 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

realgsong, 2025-04-08:
uclibc does not support fenv, so there is a problem that libraries such as opencv-dnn that require that function cannot be built. I wish rockchip would support musl libc or glibc in addition to uclibc.