LuckfoxTECH/luckfox-pico #53: The IP addresses assigned to RNDIS are -not- in RFC 1918 ranges

PuceBaboon, 2023-12-28:
The default address of 172.32.0.93 is _not_ within the "private" ranges defined
  by the IANA.  In fact this address belongs to T-Mobile USA, Inc. and is assigned as
  NetName TMO9 by them (checked in the "WhoIs" database).

   I would suggest moving this to 172.31.0.93 to bring it into the safe range.

------
> The Internet Assigned Numbers Authority (IANA) has reserved the
> following three blocks of the IP address space for private internets:
> 
>      10.0.0.0        -   10.255.255.255  (10/8 prefix)
>      172.16.0.0      -   172.31.255.255  (172.16/12 prefix)
>     192.168.0.0     -   192.168.255.255 (192.168/16 prefix)

luckfox-eng33, 2024-01-03:
hi
Thank you for your feedback, we will move it in the future

andrethomas, 2024-03-08:
@PuceBaboon For the moment you can edit the IP address as follows:

Before building: `sysdrv/tools/board/android-tools/S90usb0config`
or
After boot:  `/etc/init.d/S90usb0config` on the device file system itself.

Change the `TARGET_IP="172.32.0.94"` to another IP address if you wish

closed 2024-03-08 by andrethomas

PuceBaboon, 2024-03-12:
Andre,

  Many thanks for taking the time to reply to this issue.   I'm sure you
realize that I didn't raise it because I don't know how to change an IP
address, but I thank you for taking the time to  explain, anyway.

  I was rather hoping that LuckFox might take just a little bit of time to
change their build (and yes, I know, all of the documentation, too) to
configure this IP correctly for future products and for all users.  To put
it bluntly, my issue was raised to let LuckFox know that they have a
(small) problem that they need to fix,not to get help for myself.  So,
unless you specifically tell me otherwise, I assume that the problem still
exists and that the issue has been closed without any actual action.

  Again, thank you for for your time.

  Best wishes,

                         -John Little-

andrethomas, 2024-03-13:
Indeed, I did consider the possibility, but posted the information anyway so that any other user can in the meantime change this. The requirement for change was noted by @luckfox-eng33 above so I am sure that they will queue it in their future changes. Thanks again.