LuckfoxTECH/luckfox-pico #55: Changing rtsp configuration

csvke, 2024-01-02:
I cannot find documentation anywhere else but it seems configuration of the rtsp streams can be changed in
`/oem/user/share/rkipc-300w.ini`
As I have a Luckfox Pico Max 256M connected to the SC3336 Cam that is 300W

And after making changes, reboot is needed.

Are there documentation as to provide more details?

luckfox-eng33, 2024-01-03:
hi
RKIPC has provided the source code. 
Please check the source code yourself. 
Rockchip does not provide more documentation.

csvke, 2024-01-04:
> hi RKIPC has provided the source code. Please check the source code yourself. Rockchip does not provide more documentation.

Where is the source code?

luckfox-eng33, 2024-01-05:
[project/app/rkipc/rkipc](https://github.com/LuckfoxTECH/luckfox-pico/tree/main/project/app/rkipc/rkipc)
and
[project/app/rkipc/rkipc/docs/Rockchip_Developer_Guide_Linux_RKIPC_CN.md](https://github.com/LuckfoxTECH/luckfox-pico/blob/main/project/app/rkipc/rkipc/docs/Rockchip_Developer_Guide_Linux_RKIPC_CN.md)

closed 2024-03-05 by andrethomas