LuckfoxTECH/luckfox-pico #59: Is there a image for Alpine Linux available?

DerPicknicker, 2024-01-06:
Hey, 

just a small question I want to use the Pico Max but for me it's not clear where I can find the compatible Linux versions for this small machine. 

In addition to the question above I want to know if I can connect a usb c hub to the USB port to use an external soundcard or WiFi dongle. 

Best regards

EngineerWill, 2024-01-08:
It is compatible with busybox by default, but we also ported Ubuntu. 
Other Linux distributions need to be ported by themselves. 
Fortunately, a user contributed a post on how to port Alpine Linux, which we also reproduced in the wiki.[Luckfox-Pico-Alpine-Linux-2](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Alpine-Linux-2)

EngineerWill, 2024-01-08:
hey，
It is compatible with busybox by default, but we also ported Ubuntu. 
Other Linux distributions need to be ported by themselves. 
Fortunately, a user contributed a post on how to port Alpine Linux, which we also reproduced in the wiki.[Luckfox-Pico-Alpine-Linux-2](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Alpine-Linux-2)

DerPicknicker, 2024-01-08:
Hey @EngineerWill , 

thank you for the fast response could you link the Ubuntu image? Or do I have to compile it on my own? A already built image would be great and makes it easier to use for simple projects

EngineerWill, 2024-01-08:
luckfox provide [image](https://drive.google.com/drive/folders/1sFUWjYpDDisf92q9EwP1Ia7lHgp9PaFS)

DerPicknicker, 2024-01-08:
Thank you for the link. And then just follow the steps on the wiki I guess. This seems like a great small project for me.

andrethomas, 2024-02-10:
I just followed the wiki and built an alpinelinux image as the one provided is missing a few things like the ntp setup which makes it difficult to use apk via ssl. I like the small footprint as a well featured distro that can run from spi-nand... I think I may build a sd card image also since apk is working well already.

closed 2024-03-05 by andrethomas