LuckfoxTECH/luckfox-pico #60: Power consumption

devova, 2024-01-14:
I have Luckfox Pico MAX, flashed the stock buildroot image, and everything worked fine. Then I decided to flash [Alpine Linux](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Alpine-Linux-1), which required connecting to  UART2 and **soldering**... Well, I soldered all pads on the left side, visually without any circuit shorts. After flashing I tried to listen this UART2 on host machine and nothing appeared. So I decided to flash back the stock buildroot image, since `adb shell` was working there.

Now I noticed that the board **started being hot** too much, and after powering up, `adb shell` disconnects after a minute or two (RNDIS Virtual Network Interface just disappeared). I have USB power consumption meter and it shows **2W** all the time, I don't know how much it should consume. Anyway, red diode that blinks once per sec or two still blinking, which is saying that the board continue to work.

I assume that my soldering wasn't right, but this behaviour is strange. Any comments on this? What is expected power consumption the board should have?

devova, 2024-01-14:
I took multitool and checked is there any circuit short on pads, al soldered pads on left side where fine, but I noticed very low resistance (only 22 Oms) on right side between pad GND(38) and 3V3 OUT (36). Which means that very high current flows between ground and 3v3 power supply. Can I change internal BEC?

luckfox-eng33, 2024-01-15:
In the normal working state of the board,
1. the power supply voltage is 5V
2. the average current is about 0.2A
3. and the average power consumption is about 1W.
According to your description, the board may have been damaged.
I'm not sure about your soldering situation. 
Welding the pads normally will not cause failure.
If possible, please provide the pictures for our analysis

devova, 2024-01-15:
[image]
[image]
[image]
[image]

Here are my pictures of not ideal welding, as you might notice the board itself become even a little bit screwed

luckfox-eng33, 2024-01-16:
Your current soldering temperature is too low. 
It is recommended to solder the pad to the normal state first. Just like the picture below
[image]
Now it is difficult for us to judge whether the damage is caused by welding

devova, 2024-01-18:
Thanks anyway, I got the main answer, power consumption. Closing this issue.

closed 2024-01-18 by devova