LuckfoxTECH/luckfox-pico #64: Audio Output

DerPicknicker, 2024-01-21:
Hi, 

just a short question: is there an audio output available via the GPIOs which can be used by the Linux os? 
I2S, PWM? Or just plug an usb soundcard in? 

Thank you in advance!

luckfox-eng33, 2024-01-22:
PWM might work, but we haven't tested it

DerPicknicker, 2024-01-22:
@luckfox-eng33 ... is there any way to get audio out of this device?

luckfox-eng33, 2024-01-22:
Bidirectional transfer of files can be provided through SAMBA, SSH and ADB
We are also planning to release a version with 3.5MM interface

DerPicknicker, 2024-01-23:
3.5 Version + wifi would be sick. Any timeframe available? Is the 3.5 jack also supported by the Linux distros? 

Are there currently any other methods for getting audio out of this board.

luckfox-eng33, 2024-01-23:
The pico and plus versions can output audio through the test point lineout. For details on how to use it, please refer to Rockchip_Developer_Guide_Linux_RV1106_ACodec.pdf
[Rockchip_Developer_Guide_Linux_RV1106_ACodec.pdf](https://github.com/LuckfoxTECH/luckfox-pico/files/14022998/Rockchip_Developer_Guide_Linux_RV1106_ACodec.pdf)

DerPicknicker, 2024-01-23:
@luckfox-eng33 .. 

Thank you for the PDF. Is this supported by the ubuntu image you provided? I mean I search for a Solution which works almost out of the box. My Main question is: Can I hook a 3.5mm Jack on the GPIO-Board and I am ready to go?

luckfox-eng33, 2024-01-23:
At this stage, the Ubuntu image can only provide basic system operations, and other functions involving external interfaces or media are temporarily unavailable.
Because RK is not ready to adapt and has not provided us with the source code, we can only do this now.

DerPicknicker, 2024-01-23:
Okay this is sad. Is there any other os available which supports audio?

luckfox-eng33, 2024-01-23:
If you want a Solution which works almost out of the box, maybe busybox is the only option

closed 2024-03-05 by andrethomas