LuckfoxTECH/luckfox-pico #66: How to erase flash on linux

AlexanderSchuetz97, 2024-01-22:
I want to boot from the SD card and want to erase the flash from my luckfox pico max so it boots from the SD.

upgrade_tool EF update.img                                                                                                                                                   :(
Using /home/xxxxx/Downloads/upgrade_tool_v2.17_for_linux/config.ini
Loading loader...
Start to erase flash...
Download Boot Start
Download Boot Success
Wait For Maskrom Start
Wait For Maskrom Success
Test Device Start
Test Device Success
Get FlashInfo Start
Get FlashInfo Success
Prepare IDB Start
Prepare IDB Success
Erase IDB Start
Erase IDB Fail

Please note that using UF command to flash the busbox image works, so its not a problem with USB cable or anything like that. I am most likely just using the command wrong...

luckfox-eng33, 2024-01-23:
Can you try restarting your computer? It seems that there is a problem with the download program.

zteb1419, 2024-02-15:
The GUI utility  in luckfox-pico/tools/linux/SocToolKit (after installing SDK) works for me.

andrethomas, 2024-03-05:
@AlexanderSchuetz97 did you manage to solve this issue?

AlexanderSchuetz97, 2024-03-05:
No I have not. I have restarted my computer as luckfox-eng33 suggested but that did nothing. I could not find the tools that zteb1319 mentioned. I would very much appreciate a .tar that contains everything I need in order to get the pico to boot from SD. Preferably even everything needed to create the SD card with ubuntu. This stuff is seriously hard to find.

AlexanderSchuetz97, 2024-03-05:
After going through everything again and finding the gui tool zteb mentioned I was able to erase the flash. But the SD card did not boot. With much less ambition this time I recreated the SD card image using the busybox image, still nothing.
I am pretty confident that if I reflash the Flash with the busybox image that would work. But my end goal is to boot ubuntu from an SD card so this would be a step back.

Via Serial I am only getting RKUART when I press the reset button now.
Do you have any idea what I could be doing wrong? I also tried inserting the SD card both ways into the board since it appears to capable of accepting the micro sd card in either direction. Would be nice if you could tell me which way the pads of the sd card should be facing at least. (Towards the board or away from the board)

luckfox-eng33, 2024-03-06:
1. If you press the reset button and do not press the boot button, the serial port only receives RKUART, which means storage medium is not recognized.
2. The contacts or pads of the SD card should be facing downward, that is, toward the board.
3. It is not recommended to burn SD cards under Ubuntu because the tools provided by RK cannot burn SD cards under Ubuntu.And we did not test with dd.
4. You can use our precompiled image first. Although it may not be able to fully use all the space, it is available.https://drive.google.com/drive/folders/1r6Ulc_crJar1entKbK7GEJSq14HXL8ao

andrethomas, 2024-03-07:
> No I have not. I have restarted my computer as luckfox-eng33 suggested but that did nothing. I could not find the tools that zteb1319 mentioned. I would very much appreciate a .tar that contains everything I need in order to get the pico to boot from SD. Preferably even everything needed to create the SD card with ubuntu. This stuff is seriously hard to find. @andrethomas

The tools @zteb1419 is referring to is part of the github repository. If you clone this repo, or download it as a zip file and extract it you will find the linux tools in the tools/linux/ folder.

zteb1419, 2024-03-07:
Sorry guys. My bad. Yesterday tried again my “solution” and it doesn’t work! Even tried using different distros and unfortunately the result was total let-down.  Foolish of me. (I emailed the guy mentioned in the file ToolsRelease.txt but no answer yet.)

andrethomas, 2024-03-08:
Ok, to be clear on what to expect... if the spi-nand is erased and there is no SD card present in the board then you will only see this output on the serial interface:

[image]

This is the same output you will get if you hold the boot button down and press reset.

To be clear on the orientation of the SD card, it is inserted in the same way as in this image, with the contacts of the SD card facing DOWN.

[image]

If you inserted the SD card and it still only outputs the RKUART text on the serial interface then the SD card is not recognised as bootable media which means either the SD card is not compatible, or there may be something wrong with the image or the way you loaded the image onto the SD card.

Hope that helps

chaguidiao, 2024-04-15:
@AlexanderSchuetz97,  SocToolkit can be entirely skipped for writing to SD card.  Just use the dd command to merge all the *.img files into single image.  Take precompiled images shared by luckfox-eng33 for example (basic luckfox pico version) and referring to RK_PARTITION_ARGS, the ad-hoc makefile below generates single full.img that can be written to sd card directly.  I have verified it works with my SanDisk 16G card.

The boot-up time is painfully slow with Ubuntu though :).  

```
# cat Makefile 

IMAGE_OUT=full.img

#blkdevparts=mmcblk1:32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),6G(rootfs),-(media)

$(IMAGE_OUT): boot.img env.img idblock.img oem.img rootfs.img uboot.img userdata.img
	cp env.img $(IMAGE_OUT)
	dd bs=1k seek=32 if=idblock.img of=$(IMAGE_OUT)
	dd bs=1k seek=`dc -e '32 512 + f'` if=uboot.img of=$(IMAGE_OUT)
	dd bs=1k seek=`dc -e '32 512 256 + + f'` if=boot.img of=$(IMAGE_OUT)
	dd bs=1k seek=`dc -e '32 512 256 32 1024 * + + + f'` if=oem.img of=$(IMAGE_OUT)
	dd bs=1k seek=`dc -e '32 512 256 32 1024 * 512 1024 * + + + + f'` if=userdata.img of=$(IMAGE_OUT)
	dd bs=1k seek=`dc -e '32 512 256 32 1024 * 512 1024 * 256 1024 * + + + + + f'` if=rootfs.img of=$(IMAGE_OUT)
```

AlexanderSchuetz97, 2024-04-15:
I will give this a try. Also I do not have that much free time lately so sorry for not always reacting and trying out things quickly. I will give this makefile a tryin hoepfully sometime in the next 2 weeks. You can be certain tho that once I have something boot from the sd card that I will close this issue and report back what I did

chaguidiao, 2024-04-26:
These work for me:
https://www.taiwantrade.com/products/detail.html?productid=1874910 
https://www.singular.com.cy/kingston-gold-flash-memory-card-16gb-uhs-i-u3-class10-microsdhc-uhs-i.html

Not that I bought from the sources above though.  Just some old cards salvaged from dash cam probably.

Speaking of RKUART error, are you sure the image was not compiled with nand as target accidentally?  Do a 'strings env.img' should return something like 'blkdevparts=mmcblk*' rather than 'mtdparts=spi-nand*'.  Here's the [ubuntu promax image](https://drive.google.com/file/d/1RAY4yYBUWwDZop9Xst4KMnxrnEAVoCIv/view?usp=sharing) you can try directly DD'ing to sd card, verified with my board.

AlexanderSchuetz97, 2024-04-28:
Thank you with your image I could get it to work.
That confirms that my method of creating the SD card was wrong.
I will try to create a image using your makefile from the files luckfox provided. 
While I doubt you have put anything spicy into the image, it never hurts to try and create it from the "offical" sources.
If I cannot manage that then I will just use your image.

Thank you very much

[image]

closed 2024-04-28 by AlexanderSchuetz97