LuckfoxTECH/luckfox-pico #67: High FPS version of pico mini+camera

gehee, 2024-01-22:
I currently have the Pico Mini and Max with the SC3336 camera. I love the form factor and it's exactly what I need,  but it's just missing one thing: 1080p high fps. 

The IMX492 does 1080p @ 120fps, and the IMX415 support 1080p @ 90fps.
Can you consider releasing a kit in the same form factor as the Pico mini with an ISP that can process this FPS (maybe an rv1126?), and a camera board in the same form factor as the SC3336 with an IMX high fps sensor?

luckfox-eng33, 2024-01-23:
Thank you for your support, your feedback has been received
We will consider trying a higher FPS camera

gehee, 2024-01-23:
Rockchip seems to provide drivers for few other cameras:
https://github.com/LuckfoxTECH/luckfox-pico/tree/main/sysdrv/source/uboot/rkbin/bin/rv11

Are you planning on selling the gc2093, gc3003, sc3338 or sc4336 for luckfox pico?

luckfox-eng33, 2024-01-31:
Maybe

digitalshamansky, 2024-02-24:
Also interested in 90FPS on any compatible sensor

NairoDorian, 2024-07-21:
ALSO really interested in a faster frame rate camera, this is what we neeed

NairoDorian, 2024-10-15:
Is there any news about other compatible cameras and higher framerates? 
Thank you

zhil, 2025-08-17:
any news on this?

Sekilsgs2, 2025-12-04:
Guys - sc3336 640x480 75fps (only this mode support!)

[sc3336_CMK-OT2119-PC1_30IRC-F16.json](https://github.com/user-attachments/files/23941027/sc3336_CMK-OT2119-PC1_30IRC-F16.json)

[sc3336.json](https://github.com/user-attachments/files/23941044/sc3336.json)

Please rename sc3336.json to sc3336.ko. Put ko to /oem/usr/ko and put sc3336_CMK-OT2119-PC1_30IRC-F16.json to /etc/iqfiles - make backup original first! And reboot board. after - can using 640x480 75fps.. :)