LuckfoxTECH/luckfox-pico #68: SC3336 FPS stuck at 25

gehee, 2024-01-23:
Hi,

I have modified the rv1106_ipc app to send an rtp h265 stream however I am having  trouble configuring the ISP/sensor for lower latency.

```
rk_param_init(rkipc_ini_path_);
rk_network_init(NULL);
rk_system_init();
int rkipc_camera_id_ = rk_param_get_int("video.source:camera_id", 0); // need rk_param_init
rk_isp_init(rkipc_camera_id_, rkipc_iq_file_path_);
int ret = rk_isp_set_frame_rate(rkipc_camera_id_, 30);
if (ret < 0 ){
  printf("rk_isp_set_frame_rate(): %s\n", strerror(errno));
  return -1;
}
```

However, when I call rk_isp_set_frame_rate to set framerate to 30, the video frame rate does not change.
Can you tell me what mode are supported SC3336? I know 1080p@30fps should be supported by the SC3336.
Are the firmware/drivers/calibration configured to support this?
Note that  I would also be interested to try 720p@60fps.

The log does show me an error:
```
RTIsp3x          12:08:42-294 {ispInitDevice     :208} sensor name = m00_b_sc3336 4-0030
RTIsp3x          12:08:42-295 {ispInitDevice     :211} sensor_index = 0
RTDeviceV4L2     12:08:42-299 {open              :138} open video name(/dev/video11)
**RTDeviceV4L2     12:08:42-300 {ispCameraInfo     :549} current device:/dev/video11 isn't compatible(cap:0x84201000) device,memoryType:4, retry:0**
```

Also, `v4l2-ctl` only shows a single resolution: `2304x1296` but no mention of framerate:
```
# v4l2-ctl --device=/dev/video11 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'UYVY' (UYVY 4:2:2)
		Size: Stepwise 32x16 - 2304x1296 with step 8/8
	[1]: 'NV16' (Y/CbCr 4:2:2)
		Size: Stepwise 32x16 - 2304x1296 with step 8/8
	[2]: 'NV61' (Y/CrCb 4:2:2)
		Size: Stepwise 32x16 - 2304x1296 with step 8/8
	[3]: 'NV21' (Y/CrCb 4:2:0)
		Size: Stepwise 32x16 - 2304x1296 with step 8/8
	[4]: 'NV12' (Y/CbCr 4:2:0)
		Size: Stepwise 32x16 - 2304x1296 with step 8/8
	[5]: 'NM21' (Y/CrCb 4:2:0 (N-C))
		Size: Stepwise 32x16 - 2304x1296 with step 8/8
	[6]: 'NM12' (Y/CbCr 4:2:0 (N-C))
		Size: Stepwise 32x16 - 2304x1296 with step 8/8

```

matthewassoc, 2025-01-07:
Digging into the driver, you can see that [30fps is supported](https://github.com/LuckfoxTECH/luckfox-pico/blob/main/sysdrv/source/kernel/drivers/media/i2c/sc3336.c#L492)... I'm currently trying to figure this one out myself. You can see in some of the sample code, [they are setting the frame rate to -1](https://github.com/LuckfoxTECH/luckfox-pico/blob/40639eccad8325a669604c95cf06ab002161cb97/media/samples/example/demo/sample_demo_vi_venc.c#L1089). I can't for the life of me figure out what setting this value does.

matthewassoc, 2025-01-07:
@luckfox-eng29 @luckfox-eng33 some direction here would be greatly appreciated :)

luckfox-eng29, 2025-01-08:
您好 sc3336在选择模式时会根据 subdev（或许是isp或者cif） 的 format 使用较好的模式，sc3336 提供两种模式仅在时钟频率和帧率上有区别，而模式选择主要是看分辨率，也就是在 sc3336_find_best_fit 函数中会自动轮询到 30 fps的模式，实际上可能因为其他错误导致使用默认的25fps的模式，这部分我会进一步确定。
```

luckfox-eng29, 2025-01-08:
In the SC3336 driver source code, two modes are provided, with the main differences being the clock frequency and frame rate. When selecting a mode, it primarily depends on the resolution. The two modes are iterated, and by default, the first mode is chosen (the second mode's selection is based on the result of `sc3336_get_reso_dist` compared with the first mode).

In `sample_demo_vi_venc.c`, the parameter `ctx->vi[1].stChnAttr.stFrameRate.s32SrcFrameRate` is eventually passed to the RK MPI library for processing. According to the RK MPI documentation, setting it to `-1` means processing will follow the default maximum frame rate without any restrictions.
[image]

matthewassoc, 2025-01-08:
Thank you for response @luckfox-eng29! Because both modes define the same width and height, it will always select 25fps. So in theory, moving the [30fps mode definition](https://github.com/LuckfoxTECH/luckfox-pico/blob/main/sysdrv/source/kernel/drivers/media/i2c/sc3336.c#L481-L497) to the first element in the [supported_modes](https://github.com/LuckfoxTECH/luckfox-pico/blob/main/sysdrv/source/kernel/drivers/media/i2c/sc3336.c#L463) array should enable 30fps?

Am I understanding correctly that the only way to enable 30fps is by editing the driver source code?

luckfox-eng29, 2025-01-10:
The 30fps mode involves the initialization sequence of the I2C transmission, and at present, the only way to start the SC3336 30fps mode is to modify the driver source code

matthewassoc, 2025-01-10:
Thanks for verifying this! Also appreciate your interaction with the luckfox community. My investigations were made much easier by reading some of your previous conversations on here :)

anti-social, 2025-11-09:
Hello. I tried to comment 25 fps mode in the kernel driver but as I can see from logs this module is not compiled. Is it distributed as a binary one?

I also tried to make the source code of the module invalid but the firmware is built without any errors.

anti-social, 2025-11-10:
I looked at the code that chooses which settings to apply and as I can see it should always choose 30 fps mode:

```
static int sc3336_get_reso_dist(const struct sc3336_mode *mode,
				 struct v4l2_mbus_framefmt *framefmt)
{
	return abs(mode->width - framefmt->width) +
	       abs(mode->height - framefmt->height);
}

static const struct sc3336_mode *
sc3336_find_best_fit(struct v4l2_subdev_format *fmt)
{
	struct v4l2_mbus_framefmt *framefmt = &fmt->format;
	int dist;
	int cur_best_fit = 0;
	int cur_best_fit_dist = -1;
	unsigned int i;

	for (i = 0; i < ARRAY_SIZE(supported_modes); i++) {
		dist = sc3336_get_reso_dist(&supported_modes[i], framefmt);
		if (cur_best_fit_dist == -1 || dist < cur_best_fit_dist) {
			cur_best_fit_dist = dist;
			cur_best_fit = i;
		}
	}

	return &supported_modes[cur_best_fit];
}
```

Both modes have exactly the same size therefore they should have the same `dist`. So due to this check `dist < cur_best_fit_dist` the first one (30 fps mode) should win.