LuckfoxTECH/luckfox-pico #69: Ubuntu image memory problem on Luckfox Pro

enesbcs, 2024-01-28:
Luckfox Pico Pro contains an RV1106G2 with 128MB RAM according to datasheet.

Downloaded Luckfox_Pico_Pro_Max_Ubuntu_SDCard.zip from the Google Drive link from the wiki and it shows only 64MB available memory after booting.

```
[    0.000000] Memory: 56184K/131072K available (3510K kernel code, 291K rwdata, 1660K rodata, 164K init, 128K bss, 7304K reserved, 67584K cma-reserv
ed)

$ free
               total        used        free      shared  buff/cache   available
Mem:           56348       12140        1900           4       42308       40368
Swap:        1048572       25344     1023228
```

The reason why i choose the 128MB RAM model was that 64MB was barely enough for anything... how can i use the full 128MB? Even an apt upgrade was torture with 64MB...

enesbcs, 2024-01-28:
I guess i have to modify "rk_dma_heap_cma=66M" to "rk_dma_heap_cma=01M" in env.img file with a hex editor before writing SD image.

luckfox-eng33, 2024-01-29:
Yes, you can set it up like this if you don't need to use the camera with other RK media modules

closed 2024-01-30 by enesbcs