LuckfoxTECH/luckfox-pico #72: High loadavg

andrethomas, 2024-02-09:
Hi,

Any ideas what causes the skewed or high load avg when booting from nand?

```
# uptime
 23:54:57 up 3 min,  load average: 5.96, 2.81, 1.09
#
```

It keeps fluctuating around 5.50 to 8.50 but I can't find any obvious source of the "load"

luckfox-eng33, 2024-02-18:
You can get references from the forums
https://forums.luckfox.com/viewtopic.php?t=90

andrethomas, 2024-02-23:
It must be something to do with the factory image, maybe 1103 vs 1106 I'm not sure - works fine if you self compile an image to run from spi-nand so closing this one.

closed 2024-02-23 by andrethomas

andrethomas, 2024-03-01:
Turns out it was one or more of the oem kernel modules, probably camera related as I do not have a camera  to connect to this board.