LuckfoxTECH/luckfox-pico #73: Ramdon MAC address

andrethomas, 2024-02-11:
Hi,

I note that for each reboot a new random MAC address is "generated?" which is problematic with DHCP reservations and although I understand its benefits from a privacy perspective it would be nice to know where to change this to be static during development to make it easier to find the device on the network each time.

Any ideas?

Thanks

zteb1419, 2024-02-12:
For Pico Plus as a temporary measure I used a _.yaml_ file (in /_etc/netplan_) like this:

network:
 version: 2
 renderer: NetworkManager
 ethernets:
   eth0:
     dhcp4: no
     addresses: [192.168.2.100/24]
     routes:
     - to: default
     - via: 192.168.2.1
     nameservers:
         addresses: [192.168.2.1]

meetroger, 2024-02-15:
I prefer a solution to generate unique mac address for each device and use dhcp.

andrethomas, 2024-02-15:
Yup, but not a new one on each boot!

luckfox-eng33, 2024-02-18:
You can get references from the forums
https://forums.luckfox.com/viewtopic.php?t=90

andrethomas, 2024-02-23:
> You can get references from the forums https://forums.luckfox.com/viewtopic.php?t=90

Taking the mac from /userdata/ethaddr.txt and using it in /etc/network/interfaces as described in the forum post, e.g.

```
auto eth0
iface eth0 inet dhcp
hwaddress ether 36:45:3C:16:35:2E
```

solves this, thanks

closed 2024-02-23 by andrethomas

meetroger, 2024-02-24:
Simply put the line into the startup script:

```
if ! grep -q "hwaddress ether" /etc/network/interfaces; then echo -e "\nauto eth0\niface eth0 inet dhcp\nhwaddress ether $MACADDR" | tee -a /etc/network/interfaces > /dev/null; fi
```