LuckfoxTECH/luckfox-pico #74: Unique ID for each device?

meetroger, 2024-02-15:
How can I read a CPU ID as the unique ID for each board?

andrethomas, 2024-02-18:
Mine is different to yours, so I guess it may be unique or randomly generated

```
# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 26.17
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                                                                     vfpd32 lpae
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : Generic DT based system
Revision        : 0000
Serial          : bb20ba70ee1a42fe
```

I wonder if the BogoMIPS can be trusted... mine is a RV1106 on a pro max...

andrethomas, 2024-02-18:
The previous info was from a sd booted image, this one is booting from spi nand... i.e. different OS image

```
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 26.17
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : Generic DT based system
Revision        : 0000
Serial          : bb20ba70ee1a42fe
```

luckfox-eng33, 2024-02-18:
Yes, you can do this via cat /proc/cpuinfo | grep Serial' to get the serial number

closed 2024-03-01 by andrethomas