LuckfoxTECH/luckfox-pico #76: alpine linux, how to mount the sd card

vjoco, 2024-02-23:
Alpine linux is nice, but any oidea how to mount the sd card on luckfox ?

andrethomas, 2024-02-23:
Actually, I must have updated kernel because if I replace the boot.img with this one in the Alpine image provided by luckfox then I can see the sd card and mount it.

[boot.zip](https://github.com/LuckfoxTECH/luckfox-pico/files/14389867/boot.zip)

vjoco, 2024-02-24:
Thanks a lot, I will try the boot image you sent. I could also make with buildroot an image to use the sd card with the plus version. 
But a big question regarding building the alpine linux image: 
 - on the tty step (4.3)  I have:
 - / # ls /dev/tt*
/dev/tty
 The luckfox ttys are not there. In the previous steps, I cannot see nothing related with the luckfox hardware neither, do I miss something?

andrethomas, 2024-02-24:
They are not clear that on 4.3 that ls is done on an actual luckfox device after booting its out of context, just proceed with step 4.3.1 since you won't see any luckfox specific hardware or devices in a docker container.

vjoco, 2024-02-24:
thanks... I think I understand now.. the hardware part is done by the sdk in step 9 
Package the Rootfs and Integrate it into the SDK for Image Building

luckfox-eng33, 2024-02-24:
Although I feel it is not appropriate to discuss this issue here, because this belongs to the community contribution section, it is difficult for us to provide more help.
Regarding this issue, I think it is a problem with your device tree.Please make sure your device tree has sdmmc turned on and there is no problem with your hardware

andrethomas, 2024-02-24:
> thanks... I think I understand now.. the hardware part is done by the sdk in step 9 Package the Rootfs and Integrate it into the SDK for Image Building

Yes, Its important to remember that you're basically trying to assemble a root filesystem that will run on the luckfox pico so its about constructing it in such a way that it has whatever is needed when the system boots on the actual device, not so much about booting in docker... the docker image is just a convenient way of "constructing" the rootfs and isnt intended to behave like a luckfox pico in any way whatsoever.

vjoco, 2024-02-24:
great, I will build it with the proper device tree, and I'll let you know. great help, thanks.

vjoco, 2024-02-25:
**DONE!** 
I described in the issue 77 how I finally built alpine. 
Then, for the sd card, I made a build with the proper dts (following https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Plus-SD). Then I built rootfs with the alpine distribution and i have:
localhost:/run/sd# ls /dev/mmcblk1* -las
     0 brw-------    1 root     root      179,   0 Jan  1 00:00 /dev/mmcblk1
     0 brw-------    1 root     root      179,   1 Jan  1 00:00 /dev/mmcblk1p1
then 
mount /dev/mmcblk1p1 /run/sd
... and It is working

 It's nice because I can probably access ext4 fs too, not only fats