LuckfoxTECH/luckfox-pico #77: alpine linux porting

vjoco, 2024-02-24:
Following the steps from
https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Alpine-Linux-2
I get a limited number of devices:
[image]
 The luckfox hardware is not there. What do I miss?

andrethomas, 2024-02-24:
You won't see RV11xx specific hardware in a generic arm7 docker container.

vjoco, 2024-02-24:
thanks! is there any docker container for luckfox? otherwise how an alpine image can be generated for it?

vjoco, 2024-02-24:
great system, great help, thanks

luckfox-eng33, 2024-02-24:
Great, it seems you have solved the problem

vjoco, 2024-02-25:
**Yes, it is working**, it took a little bit more. I had to:

add the absolute path for RK_CUSTOM_ROOTFS direcltly into build.sh:
function __PACKAGE_ROOTFS()
{
local rootfs_tarball _target_dir _install_dir
RK_CUSTOM_ROOTFS="/home/pi/luckfox/luckfox-pico/sysdrv/custom_rootfs/alpine.tar.gz"
...
somehow the script didn't got the symbols from exports

then I had to remove in the docker machine any tty1 tty2 etc respawns from /etc/inittab

next step: make alpine working with the luckfox mini a, I'll keep the community updated

thanks again luckfox-eng33

vjoco, 2024-02-25:
One more comment... If a change is done in alpine docker, it's enough to run "./build.sh firmware", it's much faster than a ./build.sh

andrethomas, 2024-02-25:
> **Yes, it is working**, it took a little bit more. I had to:
>
> add the absolute path for RK_CUSTOM_ROOTFS direcltly into build.sh: function __PACKAGE_ROOTFS() { local rootfs_tarball _target_dir [...]

If you rerun build.sh lunch then the `.BoardConfig.mk` file is overwritten and you need to re-apply the change as per the steps by adding:

```
# Configure custom image directory
export RK_CUSTOM_ROOTFS=../sysdrv/custom_rootfs/alpine.tar.gz
```

Probably a more ideal way of doing it would be to create a new configuration file in `./project/cfg/BoardConfig_IPC` so that it becomes an option for `build.sh lunch` without overwriting the existing options.

closed 2024-03-01 by andrethomas