LuckfoxTECH/luckfox-pico #79: free memory (RAM), mini-a version

vjoco, 2024-03-01:
why do I see less than 64 Mb of RAM (mini-a version), any idea?

´´´
luckwifi:~# free
              total        used        free      shared  buff/cache   available
Mem:          33588       12620        8180         100       12788       15788
Swap:             0           0           0
´´´

vjoco, 2024-03-01:
would it be possible to free some allocated to gpu or similar unused hardware?

andrethomas, 2024-03-01:
It is likely you have RAM allocated in your `BoardConfig` file using `export RK_BOOTARGS_CMA_SIZE="24M"` which is in this case reserved for camera use, but according to the [SDK Environment Deployment Documentation/Wiki](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-SDK/) you can free that us for the system by changing it to 1M if you're not using camera or other hardware that uses this memory.

vjoco, 2024-03-01:
THANKS, SOLVED I've changed to 
export RK_BOOTARGS_CMA_SIZE="1M"

[    0.000000] Memory: 57068K/65536K available (4000K kernel code, 301K rwdata, 1836K rodata, 164K init, 133K bss, 7444K reserved, **1024K cma-reserved**)

free cmd give
luckwifi:~# free
              total        used        free      shared  buff/cache   available
Mem:          57232       10440       32992         100       13800       42892
Swap:             0           0           0

I HAD TO BUILD AND FLASH ONLY THE ENV, VERY FAST

closed 2024-03-01 by andrethomas