LuckfoxTECH/luckfox-pico #80: How to implement the OTA upgrade?

meetroger, 2024-03-03:
Is there any instruction to implement te OTA upgrade?

At least, I would like to upgrade the oem part (ubi4).

andrethomas, 2024-03-05:
There doesn't seem to be a way to directly perform an OTA upgrade. You will need to implement your own method for initiating, transferring and installing an update by either providing a web ui to do this through or some form of back end scripting.

vjoco, 2024-03-08:
Hi,
I use alpine linux with the pico. The package mtd-utils-ubi contains the tools necessary to attach to the volumes (/dev/mtdX). The the new ubiX_0 can be mounted rw.... so changes are possible.
 Are you you using the linux generated by buildroot? what's the output of  "mount" command?

i have for mount:
/dev/ubi1_0 on /oem type ubifs (rw,relatime,assert=read-only,ubi=1,vol=0)

vjoco, 2024-03-09:
another questions: how OTA should be performed? network (scp, http, ?) or adb? or other way (serial). 
is it just sending another oem.bin? an issue that I see is that the device may lack of temporary 30Mb to store the new oem.bin. If there is space on root mount, we can think how to unmount and deattach from the mounted oem, and writting the new one, then reboot,

kawapiki, 2024-06-04:
Any update solution which can work remotely without accessing the physical device, and works onboard memory not with SD card?

nichonaugle, 2025-06-24:
https://github.com/LuckfoxTECH/luckfox-pico/issues/153#issuecomment-2306768981

Doesn't seem that they have OTA tooling available that is supported. Best bet for non prod environments would be to just use some kind of container manager (ie Docker). 

For prod environments you would probably want to integrate an update system like RAUC however that requires modifications to u-boot and changing the partitioning. This is easier on eMMC/SD based platforms however NAND is cheaper per GB and has a smaller footprint. Regardless, implementing automatic OTA updates at the OS level is probably not worth the complexity for most developer apps.

Their platform for "building an OS" is useful for generating the download.bin, idblock, and other relevant rockchip artifacts however I have a preference for generating the ROOTFS, DTB, and KERNEL myself. Hope this helps!