LuckfoxTECH/luckfox-pico #82: Luckfox Pico Max CPU clock

shodan8192, 2024-03-08:
Hello,
My board is Luckfox Pico Max with RV1106G3 which is advertised to have 1.2GHz CPU clock.

From `/sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies` we can see that available frequences are :
`408000 600000 816000 1104000 1200000 1296000 1416000 1512000 1608000` which is inline with operating points from dts file.

During CPU 100% load, `scaling_cur_freq` shows 1608000, which seems to not be real frequency. CPU at freqs `1200000 1296000 1416000` have the same performance as at `1104000`, on `1512000 1608000` there is a small drop of it (about 0.5-1%).

Dts file for RV1103 is based on RV1106 and deletes those opps :
```
&cpu0_opp_table {
	/delete-node/ opp-1200000000;
	/delete-node/ opp-1296000000;
	/delete-node/ opp-1416000000;
	/delete-node/ opp-1512000000;
	/delete-node/ opp-1608000000;
};
```

Maybe also on RV1106 they shouldn't be here ?

luckfox-eng33, 2024-03-19:
RV1103 and RV1106 are the same chip to a certain extent, but due to different packaging and positioning issues, to a large extent their programs and settings are interoperable.

shodan8192, 2024-03-21:
I've investigated RV1106 clock behavior and here are my findings.

We have following available CPU frequencies :
```
cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
408000 600000 816000 1104000 1200000 1296000 1416000 1512000 1608000
```
Measured real clock for each and Mandelbrot benchmark execution time are as follow :
```
 set    :  real    :   bench
408000  : 407  MHz : 17830.77 ms
600000  : 599  MHz : 12110.16 ms
816000  : 1304 MHz : 5548.79  ms
1104000 : 1304 Mhz : 5548.84  ms
1200000 : 1304 Mhz : 5548.13  ms
1296000 : 1304 Mhz : 5547.74  ms
1416000 : 1304 Mhz : 5548.24  ms
1512000 : 1304 Mhz : 5546.96  ms
1608000 : 1304 Mhz : 5546.37  ms
```

The cause is that for setpoint `816000` and higher CPU clock source is **CPU_PVTPLL** as can be seen in :
https://github.com/LuckfoxTECH/luckfox-pico/blob/main/sysdrv/source/kernel/drivers/clk/rockchip/clk-rv1106.c#L156-L203

which has frequency of **1300MHz** :
https://github.com/LuckfoxTECH/luckfox-pico/blob/main/sysdrv/source/kernel/arch/arm/boot/dts/rv1106.dtsi#L68-L74

Changing there `clock-frequency` has no effect, probably this PLL is based on hardware constant length ring generator with fixed frequency.

Real frequency was measured with `arm_freq` program, using ARM `PMU`, which require kernel driver to enable counter control/read from userspace. For those who are interested, source code of `arm_freq`, `mandelbrot` and `pmu_user.ko` kernel driver as well as statically compiled executables can be found at :
https://github.com/shodan8192/RV1106_clk

PJ5168, 2024-05-02:
Even at full load stress test or coremark test, the 1.3Ghz RV1106 is lower than 60C.
Should be fine to run RV1106G3 at 1.6Ghz if applicable.
Looking forward to it!