LuckfoxTECH/luckfox-pico #86: Failed to build ninja

KOUKILEO, 2024-03-12:
I tried to install numpy on Luckfox Pico Mini B but he told me that ninja was missing, but I could not install it successfully.

pico@luckfox:~$ pip install ninja
Defaulting to user installation because normal site-packages is not writeable
Collecting ninja
  Using cached ninja-1.11.1.1.tar.gz (132 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: ninja
  Building wheel for ninja (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for ninja (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [109 lines of output]
      /tmp/pip-build-env-rd5c9qna/overlay/local/lib/python3.10/dist-packages/setuptools_scm/git.py:308: UserWarning: git archive did not support describe output
        warnings.warn("git archive did not support describe output")
      /tmp/pip-build-env-rd5c9qna/overlay/local/lib/python3.10/dist-packages/setuptools_scm/git.py:327: UserWarning: unprocessed git archival found (no export subst applied)
        warnings.warn("unprocessed git archival found (no export subst applied)")

      --------------------------------------------------------------------------------
      -- Trying 'Ninja' generator
      --------------------------------
      ---------------------------
      ----------------------
      -----------------
      ------------
      -------
      --
      Not searching for unused variables given on the command line.
      CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
      -- Configuring incomplete, errors occurred!
      See also "/tmp/pip-install-upeskyci/ninja_deb4159fe65b4e8b9b2d5baf68e909e8/_cmake_test_compile/build/CMakeFiles/CMakeOutput.log".
      --
      -------
      ------------
      -----------------
      ----------------------
      ---------------------------
      --------------------------------
      -- Trying 'Ninja' generator - failure
      --------------------------------------------------------------------------------

      --------------------------------------------------------------------------------
      -- Trying 'Unix Makefiles' generator
      --------------------------------
      ---------------------------
      ----------------------
      -----------------
      ------------
      -------
      --
      Not searching for unused variables given on the command line.
      -- The C compiler identification is GNU 11.4.0
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - failed
      -- Check for working C compiler: /usr/bin/cc
      -- Check for working C compiler: /usr/bin/cc - broken
      CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCCompiler.cmake:69 (message):
        The C compiler

          "/usr/bin/cc"

        is not able to compile a simple test program.

        It fails with the following output:

          Change Dir: /tmp/pip-install-upeskyci/ninja_deb4159fe65b4e8b9b2d5baf68e909e8/_cmake_test_compile/build/CMakeFiles/CMakeTmp

          Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_0f354/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_0f354.dir/build.make CMakeFiles/cmTC_0f354.dir/build
          gmake[1]: Entering directory '/tmp/pip-install-upeskyci/ninja_deb4159fe65b4e8b9b2d5baf68e909e8/_cmake_test_compile/build/CMakeFiles/CMakeTmp'
          Building C object CMakeFiles/cmTC_0f354.dir/testCCompiler.c.o
          /usr/bin/cc    -o CMakeFiles/cmTC_0f354.dir/testCCompiler.c.o -c /tmp/pip-install-upeskyci/ninja_deb4159fe65b4e8b9b2d5baf68e909e8/_cmake_test_compile/build/CMakeFiles/CMakeTmp/testCCompiler.c
          Linking C executable cmTC_0f354
          /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_0f354.dir/link.txt --verbose=1
          /usr/bin/cc -rdynamic CMakeFiles/cmTC_0f354.dir/testCCompiler.c.o -o cmTC_0f354
          /usr/bin/ld: cannot find -lgcc: No such file or directory
          /usr/bin/ld: cannot find -lgcc: No such file or directory
          collect2: error: ld returned 1 exit status
          gmake[1]: *** [CMakeFiles/cmTC_0f354.dir/build.make:99: cmTC_0f354] Error 1
          gmake[1]: Leaving directory '/tmp/pip-install-upeskyci/ninja_deb4159fe65b4e8b9b2d5baf68e909e8/_cmake_test_compile/build/CMakeFiles/CMakeTmp'
          gmake: *** [Makefile:127: cmTC_0f354/fast] Error 2

        CMake will not be able to correctly generate this project.
      Call Stack (most recent call first):
        CMakeLists.txt:3 (ENABLE_LANGUAGE)

      -- Configuring incomplete, errors occurred!
      See also "/tmp/pip-install-upeskyci/ninja_deb4159fe65b4e8b9b2d5baf68e909e8/_cmake_test_compile/build/CMakeFiles/CMakeOutput.log".
      See also "/tmp/pip-install-upeskyci/ninja_deb4159fe65b4e8b9b2d5baf68e909e8/_cmake_test_compile/build/CMakeFiles/CMakeError.log".
      --
      -------
      ------------
      -----------------
      ----------------------
      ---------------------------
      --------------------------------
      -- Trying 'Unix Makefiles' generator - failure
      --------------------------------------------------------------------------------

                      ********************************************************************************
                      scikit-build could not get a working generator for your system. Aborting build.

                      Building Linux wheels for Python 3.10 requires a compiler (e.g gcc).
      But scikit-build does *NOT* know how to install it on ubuntu

      To build compliant wheels, consider using the manylinux system described in PEP-513.
      Get it with "dockcross/manylinux-x86" docker image:

        https://github.com/dockcross/dockcross#readme

      For more details, please refer to scikit-build documentation:

        http://scikit-build.readthedocs.io/en/latest/generators.html#linux

                      ********************************************************************************
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for ninja
Failed to build ninja
ERROR: Could not build wheels for ninja, which is required to install pyproject.toml-based projects

vjoco, 2024-03-12:
are you using an ubuntu image?

KOUKILEO, 2024-03-12:
> are you using an ubuntu image?

That's right. I use the one from the official link: Luckfox_Pico_Mini_Ubuntu_SDCard

luckfox-eng33, 2024-03-13:
Numpy is not supported yet, the same is true in buildroot, because the toolchain does not support it.

KOUKILEO, 2024-03-13:
> Numpy is not supported yet, the same is true in buildroot, because the toolchain does not support it.

What a shame... I bought this product to run LIBSVM. Hope your support comes soon.

vjoco, 2024-03-13:
@KOUKILEO , @luckfox-eng33 

I'm more biased for alpine linux. I made a quick try in the alpine docker container, and numpy works perfectly. I don't have today a  luckfox pico with me to test it on target, but 99.999 that will work.
 This is what I've done in alpine:
 ```bash
apk add python3
apk add py3-numpy
```
then I run one of the numpy examples - numpy_test1.py
 ```python
# Importing the NumPy library with an alias 'np'
import numpy as np

# Creating a NumPy array 'nums' with provided values, including NaN (Not a Number)
nums = np.array([[3, 2, np.nan, 1],
                 [10, 12, 10, 9],
                 [5, np.nan, 1, np.nan]])

# Printing a message indicating the original array 'nums'
print("Original array:")
print(nums)

# Printing a message indicating finding the missing data (NaN) in the array using np.isnan()
# This function returns a boolean array of the same shape as 'nums', where True represents NaN values
print("\nFind the missing data of the said array:")
print(np.isnan(nums))
```
and everythink works, I get:
 ```bash
 python numpy_test1.py
Original array:
[[ 3.  2. nan  1.]
 [10. 12. 10.  9.]
 [ 5. nan  1. nan]]

Find the missing data of the said array:
[[False False  True False]
 [False False False False]
 [False  True False  True]]
 ```
 I'm surprised that ubuntu don't have the compiled numpy for armv7 arch, are you sure?
This opens another improvement, how to use docker to make an ubuntu container and transfer it to luckfox.

Anyway, tomorrow afternoon I should confirm that it is working on target.

vjoco, 2024-03-13:
IT IS WORKING on target, I have one with alpine I could connect with ssh:
 This is the bash log:
```bash
lucky:~# apk add python3
fetch http://dl-cdn.alpinelinux.org/alpine/v3.19/main/armv7/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.19/community/armv7/APKINDEX.tar.gz
(1/13) Installing libbz2 (1.0.8-r6)
(2/13) Installing libexpat (2.6.0-r0)
(3/13) Installing libffi (3.4.4-r3)
(4/13) Installing gdbm (1.23-r1)
(5/13) Installing xz-libs (5.4.5-r0)
(6/13) Installing mpdecimal (2.5.1-r2)
(7/13) Installing libpanelw (6.4_p20231125-r0)
(8/13) Installing readline (8.2.1-r2)
(9/13) Installing sqlite-libs (3.44.2-r0)
(10/13) Installing python3 (3.11.8-r0)
(11/13) Installing python3-pycache-pyc0 (3.11.8-r0)
(12/13) Installing pyc (3.11.8-r0)
(13/13) Installing python3-pyc (3.11.8-r0)
Executing busybox-1.36.1-r15.trigger
OK: 60 MiB in 71 packages
lucky:~# apk add py3-numpy
(1/4) Installing libgfortran (13.2.1_git20231014-r0)
(2/4) Installing openblas (0.3.25-r0)
(3/4) Installing py3-numpy-pyc (1.25.2-r0)
(4/4) Installing py3-numpy (1.25.2-r0)
OK: 88 MiB in 75 packages
```
Then I execute the same python test file
```bash
python testnumpy.py
Original array:
[[ 3.  2. nan  1.]
 [10. 12. 10.  9.]
 [ 5. nan  1. nan]]

Find the missing data of the said array:
[[False False  True False]
 [False False False False]
 [False  True False  True]]
```
 Main message: you can use numpy, but not installed with pip, but with apk (or apt in ubuntu)

 Does this solve your issue? I hope I could help.

nopnop2002, 2024-06-26:
The default installed g++ 11 is broken, so you need to install g++ 12 and switch to this version.
```
$ sudo apt install g++-12

$ sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 11

$ sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-12 12

$ g++ --version
g++ (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

B3B3K, 2025-02-03:
> [@KOUKILEO](https://github.com/KOUKILEO)
> 
> > That's right. I use the one from the official link: [Luckfox_Pico_Mini_Ubuntu_SDCard](https://drive.google.com/drive/folders/1sFUWjYpDDisf92q9EwP1Ia7lHgp9PaFS?usp=drive_link)
> 
> ```
>       -- Check for working C compiler: /usr/bin/cc
>       -- Check for working C compiler: /usr/bin/cc - broken
>       CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCCompiler.cmake:69 (message):
>         The C compiler
> 
>           "/usr/bin/cc"
> 
>         is not able to compile a simple test program.
> ```
> 
> The default installed C/C++ compiler is broken, so you need to install g++ 9 and switch to this version. **A 16GB or larger SD card is required for SWAP space.** **RNDIS network sharing must be enabled.**
> 
> ```
> 
> $ sudo mkswap /dev/mmcblk1p8
> Setting up swapspace version 1, size = 54726455296 bytes
> 
> $ sudo swapon /dev/mmcblk1p8
> [  105.830584] Adding 53443804k swap on /dev/mmcblk1p8.  Priority:-2 extents:1 across:53443804k SS
> 
> $ free -h
>                total        used        free      shared  buff/cache   available
> Mem:            33Mi        13Mi       2.0Mi       0.0Ki        17Mi        16Mi
> Swap:          8.7Gi        15Mi       8.7Gi
> 
> $ sudo apt update
> 
> $ sudo apt install g++-9
> 
> $ sudo apt --reinstall install libc6-dev
> 
> $ sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 1
> 
> $ sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-9  2
> 
> $ sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 1
> 
> $ sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-9 2
> 
> $ gcc --version
> gcc (Ubuntu 9.5.0-1ubuntu1~22.04) 9.5.0
> Copyright (C) 2019 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> $ g++ --version
> g++ (Ubuntu 9.5.0-1ubuntu1~22.04) 9.5.0
> Copyright (C) 2019 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> $ python3 -m pip install ninja
> ```

well, it looks well. I build numpy on buildroot buildmenu but it did not worked eathier (as expected lol). I will try ubuntu but i m scared about cant using npu and camera. Are they work?