LuckfoxTECH/luckfox-pico #88: 如何移除buildroot的软件包

WolfGenerals, 2024-03-15:
我在使用Luckfox-Pico-Plus时，SPI NAND FLASH空间不足。
经过排查发现，我不需要使用Python3.11占据了大量空间。
但是我不管怎么尝试，都无法将Python从构建的镜像中移除。

vjoco, 2024-03-15:
If you use buildroot, deselect python3 from the buildroot config
by default, it is checked!
[image]

check: https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-SDK

When I use Luckfox-Pico-Plus, the SPI NAND FLASH space is insufficient.
After investigation, I found that I don't need to use Python3.11 to occupy a lot of space.
But no matter what I try, I can't remove Python from the built image.

WolfGenerals, 2024-03-15:
我确实这么做了，但是没效果

andrethomas, 2024-03-15:
I also copy the `sysdrv/source/buildroot/buildroot-2023.02.6/configs/luckfox_pico_defconfig` file into `sysdrv/tools/board/buildroot/` otherwise if you do a clean rebuild the `luckfox_pico_defconfig` file gets overwritten with the one from the board/buildroot/ folder.

WolfGenerals, 2024-03-16:
> After `make savedefconfig` I also copy the `sysdrv/source/buildroot/buildroot-2023.02.6/configs/luckfox_pico_defconfig` file into `sysdrv/tools/board/buildroot/` otherwise if you do a clean rebuild the `luckfox_pico_defconfig` file gets overwritten with the one from the board/buildroot/ folder.

解决了。
复制`luckfox_pico_defconfig`后，运行`./build.sh clean`并重新构建，就可以移除软件包。

closed 2024-03-16 by WolfGenerals