LuckfoxTECH/luckfox-pico #89: using 256MB spi flash with minia

vjoco, 2024-03-18:
I'm trying to use
beside setting 
RK_PARTITION_CMD_IN_ENV="256K(env),256K@256K(idblock),512K(uboot),4M(boot),30M(oem),10M(userdata),200M(rootfs)"
in boardconfig,
what else has to be done?

-- right now I have
```
lucky:~# df -h
Filesystem                Size      Used Available Use% Mounted on
ubi0:rootfs              64.8M     13.0M     51.8M  20% /
```

vjoco, 2024-03-18:
ubinfo gives
```
lucky:~# ubinfo
UBI version:                    1
Count of UBI devices:           1
UBI control device major/minor: 10:126
Present UBI devices:            ubi0
lucky:~# ubinfo -a
UBI version:                    1
Count of UBI devices:           1
UBI control device major/minor: 10:126
Present UBI devices:            ubi0

ubi0
Volumes count:                           1
Logical eraseblock size:                 126976 bytes, 124.0 KiB
Total amount of logical eraseblocks:     1600 (203161600 bytes, 193.7 MiB)
Amount of available logical eraseblocks: 960 (121896960 bytes, 116.2 MiB)
Maximum count of volumes                 128
Count of bad physical eraseblocks:       0
Count of reserved physical eraseblocks:  40
Current maximum erase counter value:     2
Minimum input/output unit size:          2048 bytes
Character device major/minor:            246:0
Present volumes:                         0

Volume ID:   0 (on ubi0)
Type:        dynamic
Alignment:   1
Size:        596 LEBs (75677696 bytes, 72.1 MiB)
State:       OK
Name:        rootfs
```
the space is there...

Total amount of logical eraseblocks:     1600 (203161600 bytes, 193.7 MiB)

Amount of available logical eraseblocks: 960 (121896960 bytes, 116.2 MiB)

vjoco, 2024-03-20:
Any ideas?

andrethomas, 2024-03-20:
I'm still waiting for my W25N04KVZEIE to arrive from winbond then I'll also be giving it a go but I don't know how long that will be so I'm following you closely. I flagged it for "help wanted" so maybe @luckfox-eng33 will share some thoughts.

luckfox-eng33, 2024-03-21:
I think you may not have updated the partition table because of residual files.
You can check whether {SDK_PATH}/output/image/.env.txt

vjoco, 2024-03-21:
.env.txt seems to be ok:
mtdparts=spi-nand0:256K(env),256K@256K(idblock),512K(uboot),4M(boot),30M(oem),10M(userdata),**200M(rootfs)**

vjoco, 2024-03-21:
Maybe usefull messages from the kernel:
```
[    0.092842] spi-nand spi2.0: 256 MiB, block size: 128 KiB, page size: 2048, OOB size: 128
[    0.094047] 7 cmdlinepart partitions found on MTD device spi-nand0
[    0.094070] Creating 7 MTD partitions on "spi-nand0":
[    0.094100] 0x000000000000-0x000000040000 : "env"
[    0.096051] 0x000000040000-0x000000080000 : "idblock"
[    0.097972] 0x000000080000-0x000000100000 : "uboot"
[    0.099888] 0x000000100000-0x000000500000 : "boot"
[    0.101777] 0x000000500000-0x000002300000 : "oem"
[    0.103897] 0x000002300000-0x000002d00000 : "userdata"
[    0.105825] 0x000002d00000-0x00000f500000 : "rootfs"
```

vjoco, 2024-03-26:
Today I flashed another system. 
line in .mk config file
export RK_PARTITION_CMD_IN_ENV="256K(env),256K@256K(idblock),512K(uboot),4M(boot),30M(oem),10M(userdata),200M(rootfs)"

./build.sh lunch 

choose the target

./build.sh env

burned env ... and:
```
luck:~# df -h
Filesystem                Size      Used Available Use% Mounted on
ubi0:rootfs             172.4M     16.1M    156.2M   9% /
devtmpfs                 10.0M         0     10.0M   0% /dev
tmpfs                     9.3M    100.0K      9.2M   1% /run
shm                      23.4M         0     23.4M   0% /dev/shm
```
the rootfs size is good! 
 --- problem solved

closed 2024-03-28 by vjoco