LuckfoxTECH/luckfox-pico #90: 1-wire sensors device tree help

vjoco, 2024-03-22:
I want to access DS18b20 1-wire temperature sensors. The kernel have support for W1_SLAVE_THERM, that includes DS18b20.

 But I have no idea how to map the pin in the device tree.... This dts extra code code gives errors:
```
onewire {
	compatible = "w1-gpio";
	gpios = <&gpio0_pa4 7 GPIO_ACTIVE_HIGH>;
	status = "enabled";
	};
```

vjoco, 2024-03-22:
I can compile the kernel if i put:
```
	onewire {
		compatible = "w1-gpio";
		gpios = <&gpio0_pa4 7 GPIO_ACTIVE_HIGH>;
		status = "enabled";
	};

and then 
&pinctrl {
//Victor
	gpio0-pa4 {
		gpio0_pa4:gpio0-pa4 {
			rockchip,pins =	<0 RK_PA4 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

```

but i see no devices ....

vjoco, 2024-03-28:
I just observed problems on kernel messages:
[    0.126144] Driver for 1-wire Dallas network protocol.
[    0.126540] OF: /onewire: could not get #gpio-cells for /pinctrl/gpio0-pa4/gpio0-pa4
[    0.126569] w1-gpio onewire: gpio_request (pin) failed
[    0.126618] w1-gpio: probe of onewire failed with error -22
---- let's see

luckfox-eng33, 2024-03-29:
I don’t have DS18B20 on hand, so I can’t test it at the moment.
I'm not sure whether it can be achieved, because the real-time performance of regular Linux systems is not very high, and the support for this kind of single bus is not very good.

kawapiki, 2024-06-03:
@vjoco Have you managed it to run?

vjoco, 2024-06-03:
@kawapiki I gave up. I switch to dht111 , they have a kernel driver example. It's quite a pitty but I dont have more time to dig in.

kawapiki, 2024-06-03:
Thanks, that was my other option too

BoAC-Tech, 2024-07-27:
I got 1-wire devices working by using this module: https://github.com/pstolarz/w1-gpio-cl/
Also enabled mentioned kernel modules and now I can runtime enable/disable w1-masters for gpio ports.

wiseholder, 2025-06-03:
I got the DS18B20 sensor working with the original kernel w1-gpio driver on my LuckFox Pico Mini board. I appended the following code in \<SDK-DIR\>/sysdrv/source/kernel/arch/arm/boot/dts/rv1103g-luckfox-pico-mini.dts:
```
/{
    onewire: onewire {
        compatible = "w1-gpio";
        gpios = <&gpio1 RK_PC0 GPIO_ACTIVE_HIGH>;
        pinctrl-names = "default";
        pinctrl-0 = <&gpio1_pc0>;
        status = "okay";
    };
};

&pinctrl {
    gpio1-pc0 {
        gpio1_pc0:gpio1-pc0 {
            rockchip,pins = <1 RK_PC0 RK_FUNC_GPIO &pcfg_pull_none>;
        };
    };
};
```
In kernel config enabled w1-gpio driver:
CONFIG_W1=y
CONFIG_W1_MASTER_GPIO=y
CONFIG_W1_SLAVE_THERM=y

And that's it! System successfully recognized sensor and now I can read temperature from /sys/bus/w1/devices/\<my_device_id\>/temperature
In my case DS18B20 connected to GPIO1_C0 pin (SPI0 disabled).