LuckfoxTECH/luckfox-pico #91: Fdisk, resize the root partition

PJ5168, 2024-03-31:
Device : Pico Max
SD : TF SanDisk Ultra 32GB
OS Image : Official Ubuntu SD
Issue : The lsblk can see all the partitions on the SD, but fdisk, resize2fs, and parted can not. How to resize the root file system? Thanks!

# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mtdblock0    31:0    0  256M  0 disk
mmcblk1     179:0    0 29.7G  0 disk
├─mmcblk1p1 179:1    0   32K  0 part
├─mmcblk1p2 179:2    0  512K  0 part
├─mmcblk1p3 179:3    0  256K  0 part
├─mmcblk1p4 179:4    0   32M  0 part
├─mmcblk1p5 179:5    0  512M  0 part /oem
├─mmcblk1p6 179:6    0  256M  0 part /userdata
├─mmcblk1p7 179:7    0    6G  0 part /
└─mmcblk1p8 179:8    0 22.9G  0 part

# fdisk /dev/mmcblk1
Welcome to fdisk (util-linux 2.37.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.
This disk is currently in use - repartitioning is probably a bad idea.
It's recommended to umount all file systems, and swapoff all swap
partitions on this disk.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0x26702e50.

# resize2fs /dev/mmcblk1
resize2fs 1.46.5 (30-Dec-2021)
resize2fs: Device or resource busy while trying to open /dev/mmcblk1
Couldn't find valid filesystem superblock.

# parted
(parted) print /dev/mmcblk1
Error: /dev/mtdblock0: unrecognised disk label
Model: Unknown (unknown)
Disk /dev/mtdblock0: 268MB
Sector size (logical/physical): 512B/512B
Partition Table: unknown

luckfox-eng33, 2024-04-01:
We have not tested this. Usually we modify the partition directly in the MK file of the SDK.

PJ5168, 2024-04-02:
Meanwhile, for 32B TF, the root is 6GB, the rest is 22.9GB on p8 but not mounted.
So, for different size of TF, have to build a new image?

PJ5168, 2024-04-02:
BTW, can not mount nor format the p8 partition.

PJ5168, 2024-04-13:
BTW, checked with Milk-V Duo, the lsblk and fdisk work.

AlexanderSchuetz97, 2024-04-29:
I would also be intrested in a way to resize the root partition of the image.
fdisk/parted and the like will most likely never work as the sd card does not use a normal partition table.
As far as I could see the env.img contains the sizes. and is on the first 32k of the sd card. I have edited this file to change the root partition form 6 to 7GB but then the chip complains (via UART) about a checksum error and doesnt boot anymore. I presume that the 32K env.img has a checksum somewhere inside of it. 

I sadly cannot use SOC toolkit at all since the images created by it did never boot at all. I manually "glued" all .img files together with the help of another user and then used dd to put it onto the sd card. This worked perfectly but I am now stuck with a 6GB root partition. While this is unfortunate, its not a dealbreaker for me, as for my use case 6GB is probably enough.

Bleach665, 2024-05-06:
I'm also couldn't resize root partition. But after changing
````export RK_PARTITION_CMD_IN_ENV="32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),6G(rootfs),-(media)"````
to 
````export RK_PARTITION_CMD_IN_ENV="32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),32G(rootfs),-(media)"````
in `project/cfg/BoardConfig_IPC/BoardConfig-***.mk` config file
root partition was fit to all free space on sd card after build and flush firmware using SocToolKit.
Ubuntu, Pico Max.

nopnop2002, 2024-06-17:
@PJ5168 

>BTW, can not mount nor format the p8 partition.

I was able to use this partition as a SWAP area, but will there be any problems?

```
$ mkswap /dev/mmcblk1p8
Setting up swapspace version 1, size = 54726455296 bytes

$ swapon /dev/mmcblk1p8
[  105.830584] Adding 53443804k swap on /dev/mmcblk1p8.  Priority:-2 extents:1 across:53443804k SS

$ free -h
              total        used        free      shared  buff/cache   available
Mem:          33.2M       24.9M        1.4M      612.0K        6.9M        4.8M
Swap:         51.0G           0       51.0G
```

AlexanderSchuetz97, 2024-06-17:
mkfs.ext4 /dev/mmcblk1p8 
did the job for me.
Then you just mount it.

geraldhuguenin, 2025-01-25:
> I'm also couldn't resize root partition. But after changing `export RK_PARTITION_CMD_IN_ENV="32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),6G(rootfs),-(media)"` to [...]

Works for me too : Ubuntu Luckfox Pico A... compiling on SDK (Ubuntu) and flashing on Windows with SoCToolKit (the last version)
Thanks Bleach655 !