LuckfoxTECH/luckfox-pico #93: -lgcc : cross compile or local compile

PJ5168, 2024-03-31:
Device : Pico Max
OS image : official Ubuntu SD, full server, unminimized
Question : The RV1106 armv7 1.3Ghz is much faster than the Pi Zero W armv6 1Ghz. Is there some way to compile locally? Thanks!
Description : 
While linking : /usr/bin/ld: cannot find -lgcc: No such file or directory

The libgcc-11-dev is installed : 
pico@luckfox:~/coremark$ sudo apt-get install libgcc-11-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libgcc-11-dev is already the newest version (11.4.0-1ubuntu1~22.04).

But no 32 bit : 
pico@luckfox:~/coremark$ sudo apt-get install lib32gcc-11-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package lib32gcc-11-dev

No corss lib available : 
pico@luckfox:~/coremark$ sudo apt-get install gcc-multilib g++-multilib
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package gcc-multilib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gcc-multilib' has no installation candidate
E: Unable to locate package g++-multilib
E: Couldn't find any package by regex 'g++-multilib'

luckfox-eng33, 2024-04-01:
Not supported at the moment

PJ5168, 2024-04-11:
Fixed by : sudo apt reinstall 7 gcc and libs.
Now, ok to build code on RV1106 !~~~

closed 2024-04-11 by PJ5168

nopnop2002, 2024-05-01:
@PJ5168 

>Fixed by : sudo apt reinstall 7 gcc and libs.

I have same problem.
I want to know how to fix it.
What is 7 gcc and libs.

PJ5168, 2024-05-02:
pico@luckfox:~$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/11/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.4.0-1ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv7-a+fp --with-float=hard --with-mode=thumb --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)

PJ5168, 2024-05-02:
"sudo apt reinstall" any packages missing.
or download the package from ubuntu site, use "dpkg" to install.
FYR : https://phoenixnap.com/kb/install-gcc-ubuntu