LuckfoxTECH/luckfox-pico #95: Code page setting

PJ5168, 2024-04-02:
Device : Pico Max
OS : official ubuntu, unminimized
Issue : How to chaneg the code page
Description : 
On Windows 10,
1. The Pi Zero W + SSH + nano : can see the Chinese characters. And the python code CAN display it on the OLED screen.
2. The Pico Max + SSH + nano : can NOT see the Chinese characters. And the python code CAN display it on the OLED screen.

Please refer to the attached file for more info. Thanks!
[image]
[image]

andrethomas, 2024-04-03:
@luckfox-eng33 is localization supported on the ubuntu image?

luckfox-eng33, 2024-04-07:
I think this is an encoding issue,
You can try to modify your own encode, or edit it through an external editor.