LuckfoxTECH/luckfox-pico #97: RV3028 RTC in kernel OK

vjoco, 2024-04-03:
More for documentation:
 to use rv3028 RTC, the dts for i2c3 should have

```
&i2c3 {
	status = "okay";
	pinctrl-0 = <&i2c3m1_xfer>;
	clock-frequency = <100000>;
	rtc@52{
      compatible = "microcrystal,rv3028";
      reg = <0x52>;
   };
};
```
and the kernel should use 
[image]

 hwclock -w after a reboot... to avoid "low voltage" messages

andrethomas, 2024-04-03:
Thanks @vjoco. All we need now is someone from @luckfox-eng33 team to create entries for all the posts you have made that have been marked documentation... unless there is a way for direct contribution to the wiki?

vjoco, 2024-04-03:
@andrethomas 
Thanks for your comment... Indeed, it would be useful for the community that we could directly contribute to the wiki! A kind of "user contributed" tab

luckfox-eng33, 2024-04-07:
Thanks @vjoco. 
I think it is a good choice to publish it on the [forum ](https://forums.luckfox.com/)first, and our staff will collect it and forward it in the wiki