LuckfoxTECH/luckfox-pico #98: WiringX library support

pavelvasev, 2024-04-09:
I tried to use GPIO according to demo at https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPIO/ and found it very slow (~20 microseconds for 1 write operation). I also tried Linux's character device interface to GPIO, it is faster but not enough (~5 microseconds for 1 write operation). The problem is that data flows via Linux abstractions, which are slow.

The WiringX library https://github.com/wiringX/wiringX/ successfuly provides access to GPIO right in user space, and is fast due to direct access from user programs to GPIO registers which this library provides. By the way the WiringX already has modules for some Rockchip models.

If LuckFox Pico project may modify this library to support RK1103/06, and this will be great. The GPIO will be fast, then.

p.s. Or at least please provide documentation about GPIO registers layout in memory, so approaches like https://elinux.org/RPi_GPIO_Code_Samples#Direct_register_access may be implemented.

PJ5168, 2024-04-12:
BTW, Milk-V Duo has WiringX.....

pavelvasev, 2024-04-14:
I tested performance of GPIO.

1. LuckFox, using in-kernel linux GPIO API gpiod_set_value (e.g. GPIO consumer interface). It gives about 4 million calls per second. Same value if call chip driver "directly", e.g. gpio_chip->set(...) function.
2. MilkV, using WiringX, e.g. access GPIO from userspace, which is much simpler than writing in-kernel driver. It gives about 1 million calls per second. This is 4 times slower. Maybe it is due to Milk'v chip at all, I do not know.

In both cases, more advanced methods are required to control GPIO, with more performance (we need 24Mhz and we get only 1-4 Mhz). I assume this is possible with Luckfox in particular, however there is a lack of documentation and examples.

PJ5168, 2024-04-15:
For fast IO and state machine control, you may leverage the RP2040 PIO block.
RP2040 can emulate a 125Mhz SPI, 4Mhz I2C, I2S, UART....
Or, you can use inline assembly of ESP32 S3 to toggle the IO pins, just like 328p inline assembly.
The good thing is, you can stack up rv1106 with RP2040 or ESP32 S3. They all have Pi Pico pin out available.
[image]

pavelvasev, 2024-04-20:
@PJ5168 thank you very much for the details on RP2040!
I hoped to achieve same thing from LucykFox, but without luck..

pavelvasev, 2024-04-20:
I reclaim the issue to support WiringX library. As I investigated it's source code, the library looks non-mature. It's code is anti-optimized, for example in https://github.com/wiringX/wiringX/blob/master/src/wiringx.c#L664 strlen is called twice. So on, so on.

closed 2024-04-20 by pavelvasev