LuckfoxTECH/luckfox-pico #101: rockit.ko recompile failed

pengphei, 2024-04-28:
通过 sysdrv/drv_ko/rockit/release_rockit-ko_rv1106_arm_asm 重新构建 rockit.ko 失败：

提示:
 ERROR: modpost: "rk_tb_mcu_is_done" undefined
ERROR: modpost: "rk_tb_client_register_cb_head" undefined

因为 asm 中依赖内核的 thunder boot 功能，但是在我们的 内核中并不启用 thunder boot 功能。
这个该如何解决？是否可以提供新的 asm 源码

luckfox-eng33, 2024-04-28:
能阐述下你什么怎么遇到这个问题的，和你的动机吗？
这一部分通常是RK直接提供

pengphei, 2024-04-28:
原厂提供的sdk基于 uclinux + buildroot 进行系统构建，具有一定的可定制化。
1. 因为一直使用 openWRT ，所以在贵司的 luckfox pico 上移植，现在系统也可以运行了（基于 glibc 的基础库）。
2. 摄像头也可以使用，基于 sc3336 ，通过 v4l2-ctrl 获取原始视频，偏绿。
3. 希望可以通过 rockit 开发我们的业务，发现 rv1103/06 基于新的 mpp 和 rockit 框架需要依赖内核的 Kmpp 和 rockit，所以需要重新编译内核模块，刚好看到有 asm 汇编的，就试一试，发现了这个问题。

luckfox-eng33, 2024-04-28:
了解
我们也是催促过RK为我们开放rv1103/06与mpp 和 rockit 的源码，这样我们可以将其移植到ubuntu中
但是RK表示暂时没有开放这一部分源码的打算，所以我们也是无能为力
如果后续RK开放有关源码，我们会将其提供在我们的github上

pengphei, 2024-04-28:
我觉得我这里还是有些问题：
1. 我自己使用的是比较新的 5.10.160 的内核，而你们提供的是 5.10.60 的内核版本。

我准备切换到你们的版本上，直接使用二进制的 ko 文件试试，不过不确定 uclinux 版本的 gcc 会不会引发差异。

smzahraee, 2024-10-22:
@pengphei did you find a way to compile rockit.ko