LuckfoxTECH/luckfox-pico #102: `sudo` and `su` don't work due to setuid bit missing

platima, 2024-05-04:
It appears that `tar` only preserves permissions on extraction when running as superuser, and does not preserve owner by default.

`build.sh` needs
`tar xf $rootfs_tarball -C $RK_PROJECT_OUTPUT`
to become
`tar -x --same-owner -f $rootfs_tarball -C $RK_PROJECT_OUTPUT`

And I also added
`chmod u+s $RK_PROJECT_PACKAGE_ROOTFS_DIR/bin/busybox`
`chmod u+s $RK_PROJECT_PACKAGE_ROOTFS_DIR/usr/bin/sudo`

platima, 2024-05-04:
note: somehow this appears to have not worked, so it's quite possible that the permissions are being changed again elsewhere in the script!

luckfox-eng33, 2024-05-06:
Please describe the problem and motivation in detail

luckfox-eng33, 2024-05-07:
Does it mean you didn't use sudo when building Ubuntu?

When using sudo, you will usually be prompted
```
sudo: /etc/sudo.conf is owned by uid 1000, should be 0
sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set
```
If you want sudo to work properly, you need to reuse sudo to build the image or modify the permissions of sudoer-related files.
```
chown root:root /usr/bin/sudo
chown root:root /etc/sudo.conf
chown root:root /usr/libexec/sudo/sudoers.so
chown root:root /etc/sudoers
chown root:root /etc/sudoers.d
chown root:root /etc/sudoers.d/README
```
But I still recommend using `sudo ./build.sh` to build Ubuntu

platima, 2024-05-07:
No I mean on the resulting image running on a Luckfox Pico.

I tried creating an image with `sudo ./build.sh` but that results in the same problem when running the output image.

luckfox-eng33, 2024-05-08:
Have you tried building ubuntu before `sudo ./build.sh`?
If so, it may be due to cache file residue.
In our tests, there is no problem using `sudo ./build.sh` to build directly.

platima, 2024-05-08:
I tried with and without. I also removed out/rootfs* in case it was the cache, and I ended up cloning the entire repo again.

I will try again next weekend with a) Github images, b) Buildroot from SDK, c) Ubuntu from SDK, all fresh, and let you know.

luckfox-eng33, 2024-05-08:
OK
I will also ask my colleagues to arrange a test later.

platima, 2024-05-10:
You are right; it works if `sudo ./build.sh`. 

Should add `whoami` check to build.sh?