LuckfoxTECH/luckfox-pico #103: Error connect to Pico-Mini-B via ssh&rtsp

aerin, 2024-05-05:
BoardConfig_IPC/BoardConfig-SPI_NAND-Buildroot-RV1103_Luckfox_Pico_Mini_B-IPC.mk

I can enter through adb and login ssh both local and outer addresses:
`
adb.exe shell                                                                                                                                  
* daemon not running; starting now at tcp:5037
* daemon started successfully
# ssh root@localhost
root@localhost's password:
# pwd
/root
# exit
Connection to localhost closed.
# ssh root@172.32.0.93
root@172.32.0.93's password:
# pwd
/root
# exit
Connection to 172.32.0.93 closed.
#
`
But if I try to ssh from host computer, I get an error. I think the issue is that some setting in /etc/ssh/sshd_config is incorrect.
Example connect from linux host at Verbose mode:
`$ ssh -v root@172.32.0.93
OpenSSH_8.9p1 Ubuntu-3ubuntu0.7, OpenSSL 3.0.2 15 Mar 2022
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug1: Connecting to 172.32.0.93 [172.32.0.93] port 22.
debug1: Connection established.
debug1: identity file /home/aerin/.ssh/id_rsa type -1
debug1: identity file /home/aerin/.ssh/id_rsa-cert type -1
debug1: identity file /home/aerin/.ssh/id_ecdsa type -1
debug1: identity file /home/aerin/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/aerin/.ssh/id_ecdsa_sk type -1
debug1: identity file /home/aerin/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /home/aerin/.ssh/id_ed25519 type -1
debug1: identity file /home/aerin/.ssh/id_ed25519-cert type -1
debug1: identity file /home/aerin/.ssh/id_ed25519_sk type -1
debug1: identity file /home/aerin/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /home/aerin/.ssh/id_xmss type -1
debug1: identity file /home/aerin/.ssh/id_xmss-cert type -1
debug1: identity file /home/aerin/.ssh/id_dsa type -1
debug1: identity file /home/aerin/.ssh/id_dsa-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.9p1 Ubuntu-3ubuntu0.7
debug1: Remote protocol version 2.0, remote software version OpenSSH_8.9p1 Ubuntu-3ubuntu0.7
debug1: compat_banner: match: OpenSSH_8.9p1 Ubuntu-3ubuntu0.7 pat OpenSSH* compat 0x04000000
debug1: Authenticating to 172.32.0.93:22 as 'root'
debug1: load_hostkeys: fopen /home/aerin/.ssh/known_hosts2: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts2: No such file or directory
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ssh-ed25519
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: SSH2_MSG_KEX_ECDH_REPLY received
debug1: Server host key: ssh-ed25519 SHA256:1K4VnJKLH/xu3UnNAM4dvjp67S5QhxPq9Qqveb4R+9s
debug1: load_hostkeys: fopen /home/aerin/.ssh/known_hosts2: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts2: No such file or directory
debug1: Host '172.32.0.93' is known and matches the ED25519 host key.
debug1: Found key in /home/aerin/.ssh/known_hosts:5
debug1: ssh_packet_send2_wrapped: resetting send seqnr 3
debug1: rekey out after 134217728 blocks
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: ssh_packet_read_poll2: resetting read seqnr 3
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey in after 134217728 blocks
debug1: get_agent_identities: bound agent to hostkey
debug1: get_agent_identities: ssh_fetch_identitylist: agent contains no identities
debug1: Will attempt key: /home/aerin/.ssh/id_rsa 
debug1: Will attempt key: /home/aerin/.ssh/id_ecdsa 
debug1: Will attempt key: /home/aerin/.ssh/id_ecdsa_sk 
debug1: Will attempt key: /home/aerin/.ssh/id_ed25519 
debug1: Will attempt key: /home/aerin/.ssh/id_ed25519_sk 
debug1: Will attempt key: /home/aerin/.ssh/id_xmss 
debug1: Will attempt key: /home/aerin/.ssh/id_dsa 
debug1: SSH2_MSG_EXT_INFO received
debug1: kex_input_ext_info: server-sig-algs=<ssh-ed25519,sk-ssh-ed25519@openssh.com,ssh-rsa,rsa-sha2-256,rsa-sha2-512,ssh-dss,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,sk-ecdsa-sha2-nistp256@openssh.com,webauthn-sk-ecdsa-sha2-nistp256@openssh.com>
debug1: kex_input_ext_info: publickey-hostbound@openssh.com=<0>
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey,password
debug1: Next authentication method: publickey
debug1: Trying private key: /home/aerin/.ssh/id_rsa
debug1: Trying private key: /home/aerin/.ssh/id_ecdsa
debug1: Trying private key: /home/aerin/.ssh/id_ecdsa_sk
debug1: Trying private key: /home/aerin/.ssh/id_ed25519
debug1: Trying private key: /home/aerin/.ssh/id_ed25519_sk
debug1: Trying private key: /home/aerin/.ssh/id_xmss
debug1: Trying private key: /home/aerin/.ssh/id_dsa
debug1: Next authentication method: password
root@172.32.0.93's password: 
debug1: Authentications that can continue: publickey,password
Permission denied, please try again.`

And I can't connect to rtsp://172.32.0.93/live/0

luckfox-eng33, 2024-05-07:
Through the log, I can't tell whether the password you entered is correct. 
We tested it in our ssh login and there was no problem.

aerin, 2024-05-07:
OK, ssh may not connect due to problem on the Ubuntu 22.04 host-pc ssh_config, but what about the rtsp?

aerin, 2024-05-07:
I have tryed change ip at the sysdrv/tools/board/android-tools/S50usbdevice and S90usb0config to matching my network settings, but nothing happing...

aerin, 2024-05-08:
Thank you very much. This is my first time working with virtual adapters and I confused myself with the addresses.
By the way, in Ubuntu, rtsp does not work in vlc, so I had to install the Flatpack version.
Now everything works.

aerin, 2024-05-08:
Now everything works.

closed 2024-05-08 by aerin