LuckfoxTECH/luckfox-pico #104: Can not use certain pins on LuckFox Pico Pro/Max

kawapiki, 2024-05-06:
Hello, i'm facing GPIO use problem on my LuckFox Pico Pro/Max.
Following the guid in your wiki https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPI, but i cant manage success using as digital IO.
Pin numbers:  GPIO1_B2_d - 42 and GPIO1_B3_u - 43
[image]

These commands are succeed:
```
ls /sys/class/gpio/ 
export       gpiochip0    gpiochip128  gpiochip32   gpiochip64   gpiochip96   unexport
# echo 43 > /sys/class/gpio/export
# ls /sys/class/gpio/ 
export       gpio43       gpiochip0    gpiochip128  gpiochip32   gpiochip64   gpiochip96   unexport
# ls /sys/class/gpio/gpio
gpio43/       gpiochip0/    gpiochip128/  gpiochip32/   gpiochip64/   gpiochip96/
# echo out > /sys/class/gpio/gpio43/direction 
# cat /sys/class/gpio/gpio43/direction 
out
# echo 0 > /sys/class/gpio/gpio43/value 
# cat /sys/class/gpio/gpio43/value 
0
# echo 1 > /sys/class/gpio/gpio43/value 
# cat /sys/class/gpio/gpio43/value 
0
```
Checked with multimeter, its always show 3.3v, under load it's about 1v. I guess it's a used serial port, how to disable it properly (cannot find any references in the .dts file)?
Thanks!

luckfox-eng33, 2024-05-13:
It is not recommended to use GPIO1_B2_d and GPIO1_B3_u. 
If you need to use it, you must first turn off `fiq_debugger` which is defined in `rv1106-amp.dtsi`

kawapiki, 2024-05-13:
> It is not recommended to use GPIO1_B2_d and GPIO1_B3_u. If you need to use it, you must first turn off `fiq_debugger` which is defined in `rv1106-amp.dtsi`

Hello, thanks your help! I didn't find definition of `fiq_debugger` line in `rv1106-amp.dtsi`, anyway grepped on the source so i found this:
https://github.com/rockchip-linux/u-boot/blob/63c55618fbdc36333db4cf12f7d6a28f0a178017/arch/arm/dts/rv1106.dtsi#L103
tried to modify `rockchip,serial-id = <2>;` to` rockchip,serial-id = <3>;` , keeping the debug interface just on other serial, but nothing changed at all.

> IO42 and IO43 on Pico Pro/Max are assigned to debug ports. DTS for Pico Pro/Max can be found here.
> 
> ```
> luckfox-pico/sysdrv/source/kernel/arch/arm/boot/dts/rv1106g-luckfox-pico-pro-max.dts
> ```
> 
> You need to comment out this, create a kernel, and use the created kernel.
> 
> ```
> &uart3 {
>     status = "okay";
>     pinctrl-names = "default";
>     pinctrl-0 = <&uart3m1_xfer>;
> };
> ```
Thanks your help too, i didint tested yet (tomorrow i will), but are you sure `&uart3` is the `uart2` on the picture?

nopnop2002, 2024-05-14:
> are you sure &uart3 is the uart2 on the picture?

I'm wrong.

IO42/43 is uart2.

kornpow, 2024-06-14:
I am going to hop in your thread because I am also having GPIO issues as well on the Pro/Max. I have followed the instructions, but I cannot properly enable the GPIO in the device tree. Kernel compiles but there is no change in GPIO activity.

`rv1106-luckfox-pico-pro-max-ipc.dtsi`
(close to top of file)
```
    gpio1pc5: gpio1pc5 {
        compatible = "regulator-fixed";
        pinctrl-names = "default";
        pinctrl-0 = <&gpio1_pc5>;
        regulator-name = "gpio1_pc5";
        regulator-always-on;
    };

    gpio1pc4: gpio1pc4 {
        compatible = "regulator-fixed";
        pinctrl-names = "default";
        pinctrl-0 = <&gpio1_pc4>;
        regulator-name = "gpio1_pc4";
        regulator-always-on;
    };

    gpio1pd2: gpio1pd2 {
        compatible = "regulator-fixed";
        pinctrl-names = "default";
        pinctrl-0 = <&gpio1_pd2>;
        regulator-name = "gpio1_pd2";
        regulator-always-on;
    };

    gpio1pd3: gpio1pd3 {
        compatible = "regulator-fixed";
        pinctrl-names = "default";
        pinctrl-0 = <&gpio1_pd3>;
        regulator-name = "gpio1_pd3";
        regulator-always-on;
    };
```

(close to bottom of file)
```&pinctrl {
	spi0 {
		spi0m0_clk: spi0m0-clk {
			rockchip,pins = <1 RK_PC1 4 &pcfg_pull_none>;
		};
		spi0m0_mosi: spi0m0-mosi {
			rockchip,pins = <1 RK_PC2 6 &pcfg_pull_none>;
		};
		spi0m0_miso: spi0m0-miso {
			rockchip,pins = <1 RK_PC3 6 &pcfg_pull_none>;
		};
		spi0m0_cs0: spi0m0-cs0 {
			rockchip,pins = <1 RK_PC0 4 &pcfg_pull_none>;
		};
	};

    gpio1-pc7 {
        gpio1_pc7:gpio1-pc7 {
            rockchip,pins = <1 RK_PC7 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

    gpio1-pc6 {
        gpio1_pc6:gpio1-pc6 {
            rockchip,pins = <1 RK_PC6 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

    gpio1-pc5 {
        gpio1_pc5:gpio1-pc5 {
            rockchip,pins = <1 RK_PC5 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

    gpio1-pc4 {
        gpio1_pc4:gpio1-pc4 {
            rockchip,pins = <1 RK_PC4 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

    gpio1-pd3 {
        gpio1_pd3:gpio1-pd3 {
            rockchip,pins = <1 RK_PD3 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

    gpio1-pd2 {
        gpio1_pd2:gpio1-pd2 {
            rockchip,pins = <1 RK_PD2 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};
};
```

So at the outset, all of this config is not in the `rv1106g-luckfox-pico-pro-max.dts` file as specified in these places:
https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPIO/
https://wiki.luckfox.com/zh/Luckfox-Pico/Luckfox-Pico-GPIO/
but instead at `rv1106-luckfox-pico-pro-max-ipc.dtsi`

build instructions
```
./build.sh lunch
... uboot, etc as specified in README: https://github.com/LuckfoxTECH/luckfox-pico/tree/main
./build.sh kernel
./build.sh firmware

# double check that update.img has been updated
ls -alh output/images
```

Now burn the microSD and boot.

Everything boots fine
I can do the sysfs method like the OP did, or I used python:
```
from periphery import GPIO
from time import sleep

group = {"A": 0*8, "B": 1*8, "C": 2*8, "D": 3*8}
bank = {1: 32, 2: 32*2, 3: 32*3}
#advanced write test
pin1 = bank[1] + (group["B"]+ 2) # GPIO1_B2_d BAD
pin2 = bank[1] + (group["B"]+ 3) # GPIO1_B3_d BAD
pin3 = None
pin4 = bank[1] + (group["C"]+ 7) # GPIO1_C7_d GOOD
pin5 = bank[1] + (group["C"]+ 6) # GPIO1_C6_d GOOD
pin6 = bank[1] + (group["C"]+ 5) # GPIO1_C5_d BAD
pin7 = bank[1] + (group["C"]+ 4) # GPIO1_C4_d BAD
pin8 = None
pin9 = bank[1] + (group["D"]+ 2) # GPIO1_D2_d BAD
pin10 = bank[1] + (group["D"]+ 3) # GPIO1_D3_d BAD
```

So basically, after doing all that, the only pins that work are the pins that already work by default, and that is shown in examples on the website. Anything else, doesnt work. The pin will be sitting at 0V, 3.3V or ~1.1V, and be unresponsive.

I also disabled all of the PWM pins, and I tried as much as I could to eliminate overlapping pin configs with UARTs, PWMs, etc.

This board is perfect for me, but I am stuck on this really simple example! Please help!

Game106, 2024-08-23:
> I am going to hop in your thread because I am also having GPIO issues as well on the Pro/Max. I have followed the instructions, but I cannot properly enable the GPIO in the device tree. Kernel compiles but there is no change in GPIO activity.
> 
> `rv1106-luckfox-pico-pro-max-ipc.dtsi` (close to top of file)
> 
> ```
>     gpio1pc5: gpio1pc5 {
>         compatible = "regulator-fixed";
>         pinctrl-names = "default";
>         pinctrl-0 = <&gpio1_pc5>;
>         regulator-name = "gpio1_pc5";
>         regulator-always-on;
>     };
> 
>     gpio1pc4: gpio1pc4 {
>         compatible = "regulator-fixed";
>         pinctrl-names = "default";
>         pinctrl-0 = <&gpio1_pc4>;
>         regulator-name = "gpio1_pc4";
>         regulator-always-on;
>     };
> 
>     gpio1pd2: gpio1pd2 {
>         compatible = "regulator-fixed";
>         pinctrl-names = "default";
>         pinctrl-0 = <&gpio1_pd2>;
>         regulator-name = "gpio1_pd2";
>         regulator-always-on;
>     };
> 
>     gpio1pd3: gpio1pd3 {
>         compatible = "regulator-fixed";
>         pinctrl-names = "default";
>         pinctrl-0 = <&gpio1_pd3>;
>         regulator-name = "gpio1_pd3";
>         regulator-always-on;
>     };
> ```
> 
> (close to bottom of file)
> 
> ```
> 	spi0 {
> 		spi0m0_clk: spi0m0-clk {
> 			rockchip,pins = <1 RK_PC1 4 &pcfg_pull_none>;
> 		};
> 		spi0m0_mosi: spi0m0-mosi {
> 			rockchip,pins = <1 RK_PC2 6 &pcfg_pull_none>;
> 		};
> 		spi0m0_miso: spi0m0-miso {
> 			rockchip,pins = <1 RK_PC3 6 &pcfg_pull_none>;
> 		};
> 		spi0m0_cs0: spi0m0-cs0 {
> 			rockchip,pins = <1 RK_PC0 4 &pcfg_pull_none>;
> 		};
> 	};
> 
>     gpio1-pc7 {
>         gpio1_pc7:gpio1-pc7 {
>             rockchip,pins = <1 RK_PC7 RK_FUNC_GPIO &pcfg_pull_none>;
> 		};
> 	};
> 
>     gpio1-pc6 {
>         gpio1_pc6:gpio1-pc6 {
>             rockchip,pins = <1 RK_PC6 RK_FUNC_GPIO &pcfg_pull_none>;
> 		};
> 	};
> 
>     gpio1-pc5 {
>         gpio1_pc5:gpio1-pc5 {
>             rockchip,pins = <1 RK_PC5 RK_FUNC_GPIO &pcfg_pull_none>;
> 		};
> 	};
> 
>     gpio1-pc4 {
>         gpio1_pc4:gpio1-pc4 {
>             rockchip,pins = <1 RK_PC4 RK_FUNC_GPIO &pcfg_pull_none>;
> 		};
> 	};
> 
>     gpio1-pd3 {
>         gpio1_pd3:gpio1-pd3 {
>             rockchip,pins = <1 RK_PD3 RK_FUNC_GPIO &pcfg_pull_none>;
> 		};
> 	};
> 
>     gpio1-pd2 {
>         gpio1_pd2:gpio1-pd2 {
>             rockchip,pins = <1 RK_PD2 RK_FUNC_GPIO &pcfg_pull_none>;
> 		};
> 	};
> };
> ```
> 
> So at the outset, all of this config is not in the `rv1106g-luckfox-pico-pro-max.dts` file as specified in these places: https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-GPIO/ https://wiki.luckfox.com/zh/Luckfox-Pico/Luckfox-Pico-GPIO/ but instead at `rv1106-luckfox-pico-pro-max-ipc.dtsi`
> 
> build instructions
> 
> ```
> ./build.sh lunch
> ... uboot, etc as specified in README: https://github.com/LuckfoxTECH/luckfox-pico/tree/main
> ./build.sh kernel
> ./build.sh firmware
> 
> # double check that update.img has been updated
> ls -alh output/images
> ```
> 
> Now burn the microSD and boot.
> 
> Everything boots fine I can do the sysfs method like the OP did, or I used python:
> 
> ```
> from periphery import GPIO
> from time import sleep
> 
> group = {"A": 0*8, "B": 1*8, "C": 2*8, "D": 3*8}
> bank = {1: 32, 2: 32*2, 3: 32*3}
> #advanced write test
> pin1 = bank[1] + (group["B"]+ 2) # GPIO1_B2_d BAD
> pin2 = bank[1] + (group["B"]+ 3) # GPIO1_B3_d BAD
> pin3 = None
> pin4 = bank[1] + (group["C"]+ 7) # GPIO1_C7_d GOOD
> pin5 = bank[1] + (group["C"]+ 6) # GPIO1_C6_d GOOD
> pin6 = bank[1] + (group["C"]+ 5) # GPIO1_C5_d BAD
> pin7 = bank[1] + (group["C"]+ 4) # GPIO1_C4_d BAD
> pin8 = None
> pin9 = bank[1] + (group["D"]+ 2) # GPIO1_D2_d BAD
> pin10 = bank[1] + (group["D"]+ 3) # GPIO1_D3_d BAD
> 
> pin1_gpio = GPIO(pin1, "out")
> pin2_gpio = GPIO(pin1, "out")
> pin4_gpio = GPIO(pin4, "out")
> pin5_gpio = GPIO(pin5, "out")
> pin6_gpio = GPIO(pin6, "out")
> pin7_gpio = GPIO(pin7, "out")
> pin9_gpio = GPIO(pin9, "out")
> pin10_gpio = GPIO(pin10, "out")
> ```
> 
> Now that Ive initialized all of the gpio pin objects I made a testing function, which tests whether or not the pin does what I ask:
> 
> ```
> def check_state(gpio_pin):
>     initial_state = gpio_pin.read()
>     gpio_pin.write(True)
>     sleep(0.25)
>     second_state = gpio_pin.read()
>     if second_state != True:
>         return False
>     gpio_pin.write(False)
>     sleep(0.25)
>     three_state = gpio_pin.read()
>     if three_state != False:
>         return False
>     return True
> 
> 
> check_state(pin10_gpio) # bad
> check_state(pin4_gpio) # good
> check_state(pin5_gpio) # good
> ```
> 
> So basically, after doing all that, the only pins that work are the pins that already work by default, and that is shown in examples on the website. Anything else, doesnt work. The pin will be sitting at 0V, 3.3V or ~1.1V, and be unresponsive.
> 
> I also disabled all of the PWM pins, and I tried as much as I could to eliminate overlapping pin configs with UARTs, PWMs, etc.
> 
> This board is perfect for me, but I am stuck on this really simple example! Please help!

I have also encountered the same problem. Have you solved it?

kawapiki, 2024-08-23:
No, switched to other board with "better" support. If you want similar performance try a milk-v product

Game106, 2024-08-23:
来信我已收到，谢谢。

luckfox-eng33, 2024-08-23:
Hi guys, if you still encounter that the IO function cannot be used, it is recommended to use the io command to determine whether there is a problem with your device tree.
Please refer to the specific io instructions[Rockchip_RV1106_User_Manual_GPIO.pdf](https://github.com/user-attachments/files/16725839/Rockchip_RV1106_User_Manual_GPIO.pdf)

kornpow, 2024-08-23:
What ended up working for me was actually booting the board with the proper software. Even though I thought I was burning the SD card properly, it was actually still booting from the factory image on the SPI flash, which is severely limited.

Now I am able to read and write to all pins thats I have attempted to use. I still haven't had any luck configuring the pull-up/down resistors.

Also consider using the `luckfox-config` utility which is installed on the images in order to configured the IOs.

johnathanhawn, 2024-12-31:
I too have the same issues with not being able to use pin 42 and 43.... tried some of the stuff above and only ended up locking myself out of gpio for user and root

anybody know how to change default user settings in the build aswell?

Game106, 2024-12-31:
来信我已收到，谢谢。