LuckfoxTECH/luckfox-pico #107: AlpineLinux image may be corrupted

nopnop2002, 2024-05-13:
The AlpineLinux image is published here.

https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Alpine-Linux-1

I downloaded it, wrote it to the SD card, and started mini-a.
The image may be corrupted.

[image]

nopnop2002, 2024-05-13:
thank you for your reply.

I understand that if I write this firmware to Mini-B, I can use USB-WiFi, but is that correct?

vjoco, 2024-05-14:
> thank you for your reply.
> 
> If I write this firmware to Mini-B, can I use USB-WiFi?

yes, but it's not "out of the box". you have to enable usb host support, and add the kernel drivers and firmware for your adapter. Which wifi adapter you you have? (lsusb gives vid and pid)

nopnop2002, 2024-05-14:
I studied alpinelinux a little bit.

Perhaps I also need to change the SSID and password of the WiFi router to suit my environment?

I have almost everything available as a USB dongle.

peterwillcn, 2024-05-22:
[image]