LuckfoxTECH/luckfox-pico #108: rndis not discoverable on apple devices( ipad, iphone)

ciostef, 2024-05-13:
I have all LuckFox boards and i am trying to do a hardware password manager/ encryptor  that can connect to the usb port, computers and phones.
Because of limitation of the Apple environment regarding to the direct connectivity to the phones/tablets, i want to connect the LuckFox boards to the iPhone/iPad USB connector directly and seen as rndis devices.

I also have Raspberry Pi boards that can do this , but i want to use your products/ boards like LuckFox mini B.

What i need to change in the configuration  for the board to be seen as rndis device?

The board is seen as rndis device properly only in windows.

luckfox-eng33, 2024-05-16:
I'm not sure whether iPhone/iPad USB requires special processing?
Our tests in windows and ubuntu are all normal.

ciostef, 2024-05-20:
My principal interest is the connectivity with iphone/ipad. What i know that raspberry pi is working as RNDIS USB, LuckFox don't at this moment.