LuckfoxTECH/luckfox-pico #109: Static IP address of the USB virtual dongle

nopnop2002, 2024-05-14:
I created a custom buildroot according to Wiki.
Everything went well.

I have two questions.

(Q1)
According to Wiki, the steps are as follows.

```
cd <SDK directory>/sysdrv/source/buildroot/buildroot-2023.02.6/
make luckfox_pico_defconfig
make menuconfig
make savedefconfig 
make

cd <SDK directory>
./build.sh lunch
./build.sh
```

Is savedefconfig necessary?

(Q2)
Static IP address of the USB virtual dongle is 172.32.0.93.

[image]

Can a custom buildroot change this?

luckfox-eng33, 2024-05-16:
1. If you use buildroot, we recommend using `./build.sh` instead of `sudo ./build.sh`
2. If you do not use automatic compilation scripts later, it is not necessary, but it is necessary if you need automatic compilation scripts `build.sh`.
3. You can check `sysdrv/tools/board/android-tools/S90usb0config` by yourself
4. Same as question 1, `sudo` is only required when building ubuntu, which involves permission issues.

nopnop2002, 2024-05-16:
@luckfox-eng33 

Thank you for answering.

closed 2024-05-16 by nopnop2002